libretro/Lakka #411: Support for Pine A64

LukasBombach, 2016-06-25:
I'm surprised I can't find a ticket for this on GitHub. From googling I see I am not the only one hoping for support for Pine A64. What is the status on this? Is it being worked on? Thanks.

Kivutar, 2016-06-25:
Hello, we received a Pine A64. I didn't try to port yet because the software part is still unstable : the sunxi documentation here doesn't even state mali http://linux-sunxi.org/Pine64

There is also the problem of arm64, most of the emulators doesn't have a 64bit dynarec for arm, so we need to compile a 32bit image, or a multilib image.

LukasBombach, 2016-06-25:
Ok I only understand half of this, but I get an idea. Thank you for clarifying. Still hoping very much for this cos this probably means N64 and others of that generation (maybe even later) in full speed :).

drracing07, 2016-11-29:
Any update on this?  I have 2 Pine64+ 2gb versions I could test with.  I'm not a programmer but am willing to learn/help out/provide any help i can.