libretro/Lakka #415: Automatically Reconnect Bluetooth Controller

cbluth, 2016-06-29:
I have an OUYA Controller (anyone wanna buy an OUYA Console?).
I run lakka on an Asus Chromebox, which has bluetooth and wireless built in.
I has setup the wireless to connect automatically once it boots, which is awesome.

I can pair the controller manually, but everytime I boot up lakka, the controller fails to connect, bummer!

> Lakka:~ # bluetoothctl 
> [NEW] Controller 54:27:1E:B6:78:3B BlueZ 5.37 [default]
> [NEW] Device B8:5A:F7:C3:8C:E1 OUYA Game Controller
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [bluetooth]# connect B8:5A:F7:C3:8C:E1
> Attempting to connect to B8:5A:F7:C3:8C:E1
> [bluetooth]# connect B8:5A:F7:C3:8C:E1
> Attempting to connect to B8:5A:F7:C3:8C:E1
> Failed to connect: org.bluez.Error.InProgress
> Failed to connect: org.bluez.Error.Failed
> [bluetooth]# trust B8:5A:F7:C3:8C:E1
> Changing B8:5A:F7:C3:8C:E1 trust succeeded
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: no
> [bluetooth]# connect B8:5A:F7:C3:8C:E1
> Attempting to connect to B8:5A:F7:C3:8C:E1
> Failed to connect: org.bluez.Error.Failed
> [bluetooth]# connect B8:5A:F7:C3:8C:E1
> Attempting to connect to B8:5A:F7:C3:8C:E1
> [CHG] Device B8:5A:F7:C3:8C:E1 Connected: yes
> [CHG] Device B8:5A:F7:C3:8C:E1 Paired: yes
> Connection successful
> [OUYA Game Controller]# trust 54:27:1E:B6:78:3B
> Device 54:27:1E:B6:78:3B not available
> [OUYA Game Controller]# quit
> [DEL] Controller 54:27:1E:B6:78:3B BlueZ 5.37 [default]

You can see that the top several lines, im doing a "quick single press" on the controller to turn it on, and it fails to connect.
Only once I use bluetoothctl to manually connect it, then it will work.

Perhaps I am missing something, but is there a way for me to fix this behavior, or is it a bug?

cbluth, 2016-06-30:
I just bought a second ouya controller to verify it fails with another controller. Also, if it does fail, I'm willing to ship it to any developer as a donation, to debug with.

Also, I have tested the bluetooth auto connect with a genuine sixaxis DS3 controller, and it works fine.

Kivutar, 2016-06-30:
Ok. Having a OUYA controller will really help us to debug. You can send it at the address on this page: http://www.lakka.tv/doc/Wishlist/

Thank you

hooby3dfx, 2016-09-11:
I got my original OUYA controller paired, and the pairing is retained across reboots etc.
I recommend trying this:
0. Turn off the OUYA controller
1. In bluetoothctl use the remove <ADDR> command to start from a clean slate
2. Run scan on
3. Press and hold the menu button on the OUYA controller until the middle 2 LEDs slow flash (pairing mode)
4. It should show up then as discovered
5. Run the pair <ADDR> command to complete the pairing

The pair command is key.

I actually have a separate issue with my OUYA controller (this is the original model with mouse trackpad).
It never shows up as something that can be selected in the Input menu Device Index.
I am wondering if its because of the mouse. 

This is from dmesg:
```
[    9.390903] Bluetooth: Core ver 2.20
[    9.391002] NET: Registered protocol family 31
[    9.391012] Bluetooth: HCI device and connection manager initialized
[    9.391033] Bluetooth: HCI socket layer initialized
[    9.391046] Bluetooth: L2CAP socket layer initialized
[    9.391093] Bluetooth: SCO socket layer initialized
[    9.399627] Bluetooth: HCI UART driver ver 2.3
[    9.399652] Bluetooth: HCI UART protocol H4 registered
[    9.399657] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    9.399663] Bluetooth: HCI UART protocol BCM registered
[   20.176813] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[   20.176871] Bluetooth: HIDP socket layer initialized
[   20.182881] hid-generic 0005:2836:0001.0001: unknown main item tag 0x0
[   20.183632] input: OUYA Game Controller as /devices/platform/soc/3f201000.uart/tty/ttyAMA0/hci0/hci0:11/0005:2836:0001.0001/input/input1
[   20.184484] hid-generic 0005:2836:0001.0001: input,hidraw0: BLUETOOTH HID v1.02 Mouse [OUYA Game Controller] on b8:27:eb:3e:ca:27
```

hooby3dfx, 2016-09-11:
OK yea that was it - I added a udev rule similar to the Nvidia Shield one to get it to appear.
Will create a PR for that and the button cfg!

Kivutar, 2016-10-04:
Can we close this?

Kivutar, 2016-10-05:
yeah I see the PR merged.

closed 2016-10-05 by Kivutar

travisghansen, 2017-03-15:
Is the udev rule missing in 2.0-rc3?  I can connect manually as well (auto reconnect isn't working), however I get no control of the interface and can't select it anywhere either..

New to lakka (and RetroArch/libretro in general) so I may have something completely off :(

Kivutar, 2017-03-15:
I see the same udev rules here https://github.com/libretro/Lakka/tree/lakka/packages/libretro/retroarch/udev.d and here https://github.com/lakkatv/Lakka-LibreELEC/tree/Lakka-V2.0/packages/libretro/retroarch/udev.d

travisghansen, 2017-03-15:
Hmm, I get nothing with it.  I mean, I can `cat /dev/input/js0` or whatever and see data flying by so I know it's 'working'...just not doing anything in the interface though.

hooby3dfx, 2017-03-16:
The fix was actually made on packages/linux/udev.d/99-fix-controller.rules
See https://github.com/libretro/Lakka/pull/482/files
That file is no longer present in the LibreELEC fork.

travisghansen, 2017-03-16:
Cool I'll hack it in there and try it out.

Kivutar, 2017-03-16:
You can add it somewhere in your storage partition.

I added it back in Lakka LE https://github.com/lakkatv/Lakka-LibreELEC/commit/1a48d4476c8d389b3116739e68ac561cf3384e86

travisghansen, 2017-03-16:
Aside from not *doing* anything it also wasn't reconnecting after a reboot.  Is something missing for that process to work as well?  I mean, I stepped through the bluetoothctl `connect` and `trust` stuff from the cli but I had to redo that after every reboot.  Am I missing something?

hooby3dfx, 2017-03-16:
What board are you using?
I noticed this also yesterday. 
See https://github.com/lakkatv/Lakka-LibreELEC/issues/83

travisghansen, 2017-03-17:
I'm running on a RPi3.

travisghansen, 2017-03-17:
Update: I've manually copied the udev file in place and can confirm that the controller actually works now.  I manually mapped the buttons as the defaults seemed pretty messed up but at least it works.

Not connecting at boot is still an issue however.

hooby3dfx, 2017-04-05:
@travisghansen I have also noticed that even if the Ouya controller is paired with the system, if it is disconnected (manually via bluetoothctl) and then turned back on, it will fail to reconnect by itself (even without a system reboot).
Can you confirm?

travisghansen, 2017-04-05:
I'm out for a bit but I should have some time to review early next week.  I'll let you know..