libretro/Lakka #416: Use Title thumbnails as a fallback for games without Boxart thumbs

markwkidd, 2016-06-29:
[I'd say this issue falls under enhancement or feature request.]

Of all of the currently compiled thumbnail graphics available, it seems like the pattern that boxart is the least complete for most playlists/emulated systems. That makes sense.

I would argue that it's not a great user experience to have thumbnails set to Boxart and then to have a lot uncertainty while navigating through the horizontal menu about whether an image appears depending on which playlist and game are selected. That said, I do like the boxart which *is* available.

My suggestion is to use the Title thumbnail as the fallback when:
1) Thumbnails are set to Boxart and
2) A game in focus on the horizontal menu 2a) has no Boxart thumbnail and 2b) it does have a Title thumbnail.

Rather than a programmatic solution, I'd suggest starting with a copy of each system's Named_Titles folder. Then simply copy in the contents of that playlist's Boxart folder, overwriting any Title thumbnails with the same file name. Over time as new Boxart becomes available for games currently without, it would be just a matter of replacing the Title substitute with the new Boxart.

markwkidd, 2016-09-02:
I've learned a lot about RetroArch thumbnails since I posted this issue. It doesn't really seem like a Lakka issue though, right?

I am wondering now if the more elegant solution would be to have a 'system default' Thumbnail that was part of the XMB theme. This would appear for any playlist item where there was not a Thumbnail matching the current Menu setting. 

But again, that seems like it would go in another repository. Should I close this issue?

Kivutar, 2016-09-13:
Yeah, it has its place in RA I think. Thanks.

closed 2016-09-13 by Kivutar

markwkidd, 2016-09-13:
Cool. This has become this: https://github.com/libretro/RetroArch/issues/3574