libretro/Lakka #417: Mupen64 core crashes

cbluth, 2016-06-30:
I am running lakka on a small Intel based 64bit system, and mupen64 core crashes at the same point. I have md5sum'd the ROM to make sure its a good dump, and its good.

Is there a way to only update a certain core, even if that way is manually?

cbluth, 2016-06-30:
Here is a verbose log: https://gist.github.com/anonymous/97daa09189dcf5baee1f97493bd8ae7f

cbluth, 2016-06-30:
Nevermind, I updated to the latest core, found here: http://buildbot.libretro.com/nightly/linux/x86_64/latest/

and placed it in /tmp/cores/ and it isnt crashing any longer.

Kivutar, 2016-06-30:
Ok this mean we need to update mupen64plus in our images. I reopen this for now and we will close it when mupen is updated and tested on all platforms.

Kivutar, 2016-07-11:
I've updated mupen since then http://static.kivutar.me/Lakka-Generic.x86_64-devel-20160710153555-r21519-gd0ce46d.img.gz

Can you try this image and confirm that the bug is gone please?

closed 2016-08-05 by Kivutar