libretro/Lakka #418: SHANWAN PS3 GamePad Bug

Rocklima2, 2016-06-30:
SHANWAN PS3 GamePad auto-config is nice on lakka but controller vibration not stop, all time on lakka menu, how disable controller vibration to force stop.you need hardware for test, how address i send.nice working with lakka.

input_driver = "udev"
input_device = "SHANWAN PS3 GamePad"
input_vendor_id = "1356"
input_product_id = "616"

Rocklima2, 2016-07-25:
@Kivutar you need controller for solve this issue? i send from aliexpress , but you have this model i try send other model to help project.

heylookltsme, 2016-09-03:
I'm also experiencing this. I don't know if it's related or not, but I'm also having trouble getting this controller set up via bluetooth.

gouchi, 2016-09-03:
Do you reproduce the issue with [latest release](http://sources.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160811142524-r21565-gc79651a.img.gz) ?

heylookltsme, 2016-09-03:
The vibrating issue, yes. It just continuously vibrates when it's plugged in via USB. 

However, I was mistaken about the connection issue. I was actually able to pair two SHANWAN PS3 controllers just fine. The issue was I had already plugged in a different USB controller and I didn't realize that only one controller at a time can access the lakka interface. Once I unplugged the wired controller, I was able to use a wireless one just fine.

Kivutar, 2016-10-02:
I could reproduce this bug.

TheDcoder, 2016-10-04:
I have a Red Gear Smartline Wired Gamepad with 2 input modes. In DirectInput mode, lakka recognizes it as `ShanWan PS3 GamePad`, However I am NOT able to reproduce this issue... Here is my output from `lsusb`:
`Bus 002 Device 007: ID 2563:0523`

Kivutar, 2016-10-04:
It looks different than mine.

TheDcoder, 2016-10-04:
My controller might be imitating a different ShanWan...

Kivutar, 2016-10-04:
So this bug affects Linux and not Windows. I was able to use the Shanwan on windows without that continuous vibration.

TheDcoder, 2016-10-05:
I tested it in Lakka and Fedora 24

Rocklima2, 2016-11-13:
Any progress?i try test other linux version to help, get information or one version working.

Rocklima2, 2016-12-03:
@Kivutar this model of SHANWAN PS3 working better for wireless because use PS3 driver controller , but with usb cable use SHANWAN driver this cause vibrating bug, to solve this model have use PS3 driver with usb cable.

rhkarls, 2016-12-13:
@Rocklima2 Did I understand correctly that you fixed the constant vibration? Would you mind posting the cfg file for the fix if so?

Rocklima2, 2016-12-24:
With usb cable plugged wrong driver is loaded , but with wireless driver loaded is correct, this change solve this bug, @rhkarls i understand o problem, i need Kivutar to solve.

mkrawiec, 2017-01-02:
I was also able to reproduce this issue. No idea how to fix. 
`Bus 001 Device 014: ID 054c:0268 Sony Corp. Batoh Device / PlayStation 3 Controller`

Udev log - http://pastebin.com/VAzgbcmu

willbelr, 2017-04-06:
@Rocklima2 Please can you share some advices about how you proceeded to bind the new driver? Thanks!

justnoob, 2017-04-18:
I was also able to reproduce this issue on Retropie 4.1; Raspbian Jessie and Arch. 

The controller works fine on bluetooth (after applying the patched version of bluez on the Pi)

weaster007, 2017-04-28:
Hey I am not really experienced, but could someone tell me how to apply the pached bluez to my Pi? Would be awesome to get the ShanWan controller to work over Bluetooth because I also have the problem with the vibration when useing usb connection.

Kivutar, 2017-04-28:
Hi, there are no patch to fix this bug yet.

weaster007, 2017-04-28:
justnoob said: 
"The controller works fine on bluetooth (after applying the patched version of bluez on the Pi)"
So there seems to be a patch, or am I wrong?

mkrawiec, 2017-04-28:
Can confirm, works fine without any patch. Just follow official instructions on setting up Bluetooth controller.

willbelr, 2017-04-28:
I had to install a patched version of bluez to get the bluetooth to work with arch linux. I just installed "bluez-ps3" from the aur. You say you got it working with the regular bluez package?

Also about the vibrating bug, I think it is kernel related so perhaps this thread should be moved?

Kivutar, 2017-04-28:
Yes it is kernel related, but we can keep the thread until we patch the kernel.

The default bluez should be enough for DS3 to work.

weaster007, 2017-04-28:
Excuse my stupid question, but does the ps3 bluetooth addon in retropie consist of bluez or is it something else? I really do not know much about the different packages... Also I think the default bluez does not support a e.g. ShanWan controller which is why I might need to patch it.

justnoob, 2017-04-28:
weaster, i had to use the [bluez patched by letzel](https://github.com/luetzel/bluez), from there, was pretty straight forward.
I was unable to pair the ps3 clone before though (Shanwan)

mkrawiec, 2017-04-28:
Did you follow the http://www.lakka.tv/doc/Wireless-Dualshock/ guide?

joaodomingos, 2017-06-22:
I got the solution for the wired controller not to rumble!
Open the controller, remove the wires connected to the rumble device, done!

Kivutar, 2017-06-22:
Sounds like castration.

TheDcoder, 2017-06-22:
LOL @Kivutar!

joaodomingos, 2017-06-22:
@Kivutar looooool yes a little, but at least it's not vibrating the whole time, and the buttons still work correctly

jzobac, 2017-06-22:
On 3.4 linux kernel no rumble bug (BananaPi Lakka version).