libretro/Lakka #419: HDMI on pc

BrokenHDMI, 2016-07-01:
Reving an old pc and i get to the bootloader (Where it says run live/install) and if i click on either of those my new 3d tv says mode not supported and my monitor and other tvs just go black. any way you can help me. Also will an orange pi work with retro pi/lakka

Kivutar, 2016-07-02:
Ok, I'm not familiar at all with 3D TVs. But here are a few questions:

What happen when you use the exact same setting, but another linux distribution?

Have you tried all the settings of your TV to disable 3D?

BrokenHDMI, 2016-07-02:
yes and ihave the same problem with other monitors and tvs. and on other linux distribution i can run it live but when i install it it goes black/mode not supported

cbluth, 2016-07-05:
@BrokenHDMI 
What happens when you use a Live USB of Ubuntu 16.04 LTS on the same hardware, does the hdmi work when you do that?

BrokenHDMI, 2016-07-06:
yes it works fine

BrokenHDMI, 2016-07-06:
Yes it works fin clbuth

Kivutar, 2016-07-09:
Could be related to https://github.com/libretro/Lakka/issues/365

jonny555, 2016-07-20:
I have lg 3d tv 2016 I'm it's more than likely your hardware motherboard or drivers or use a different hdmi not 4k 3d arc one

Kivutar, 2016-08-12:
Any news on this? Did you try another cable?

Kivutar, 2016-10-04:
No news, I close

closed 2016-10-04 by Kivutar