libretro/Lakka #420: (Rpi3) Stuck on rainbow screen

natinusala, 2016-07-02:
I tried building Lakka myself using the information you provide on the wiki. It worked the first time, but when I tried to update RetroArch to its latest version, I am now stuck at the rainbow screen when booting. The weird thing is that even when reverting my changes and putting the original RA shipped with Lakka, it does the same. Raspbian works fine on the other hand.

The `kernel.img` file is present on the FAT32 partition and weights approx. 5mb.

(I removed the `disable_splash=1` of the config file to see the rainbow screen)

Kivutar, 2016-07-02:
Being stuck at the rainbow screen means that the bootloader or the kernel have a problem. How did you compile?

natinusala, 2016-07-02:
Using the exact command that you provide on the wiki : 

DISTRO=Lakka PROJECT=RPi2 ARCH=arm make image

Kivutar, 2016-07-02:
Let try to reflash many times?

natinusala, 2016-07-02:
I already tried 3 times, the last one with a complete fresh copy of the repo

Kivutar, 2016-07-03:
Sorry I don't know how to help you. If it worked the first time, and doesn't work the second time, even with a fresh clone..

Are you sure you've not forget any obvious thing like gunziping the image before flashing it? It happened to me many times.

Also, try another SD card.

natinusala, 2016-07-03:
Upon formatting with SDFormatter on Windows and flashing back the image using dd on Linux, there are two partitions on the card : 

One is of type fat16, has 256mb of space and contains the "classic" raspberry pi fat partition content (config, kernel, boot image...)
The other one is of type ext4, has only 34mb and contains a ".please-resize-me" file and a ".lost+found" folder.
The rest if nothing but 7gb of unallocated space 

My guess is that dd doesn't work well, at least for me. Isn't there another tool that I could use to flash it on Linux ?

natinusala, 2016-07-03:
Okay, I got it to work. I took care to unmount all the sdbX partitions before flashing the drive. You should maybe add this to the guide ?

closed 2016-07-03 by Kivutar