libretro/Lakka #421: Broken CHD support in Lakka's MAME 2003 core

markwkidd, 2016-07-02:
Attempting to load any games which use a CHD results in this error:
`blitz.chd NOT FOUND`
`ERROR: required files are missing, the game cannot be run.`
`RetroArch [ERROR] :: Failed to load content.`

This error isn't present in the version of the MAME 2003 core currently available from the buildbot. 

The issue I'm describing started as a mame2003-libretro issue, which turned out to be a combination of this problem and a more tricky regression in CHD support. If you are interested in that issue: https://github.com/libretro/mame2003-libretro/issues/38

Kivutar, 2016-07-02:
What if I update mame2003? Would that fix the problem you think?

markwkidd, 2016-07-02:
I believe so.  Whatever version of that core the buildbot is making for
Linux x64 seemed to fix the issue and be otherwise working well.  I just
did quick spot checking though.

<https://github.com/libretro/Lakka/issues/421#issuecomment-230114821>, or mute
the thread

Kivutar, 2016-07-10:
Try this build, it have an updated mame2003 http://static.kivutar.me/Lakka-Generic.x86_64-devel-20160710153555-r21519-gd0ce46d.img.gz

markwkidd, 2016-07-14:
Thank you! I tried that build and it is loading CHDs correctly. That seems to have resolved this issue.

closed 2016-07-14 by Kivutar