libretro/Lakka #422: MAME manual Screenshots very small file size not working ?

MopheusDG, 2016-07-02:
Hi, I've been checking an issue and not sure why is it happening. I have a collection of MAME games, including games like "clowns" and "bezerk", and only those games PNG screenshots do not show. All my other roms sshots work fine and the only diference I found is those PNG files are really small (cause games are old and most of the screen is black I suppose). Of course those SSHOTS work on MAME for Windows, etc.

Can anyone confirm this ?

Thanks.

markwkidd, 2016-07-31:
Hey @MopheusDG -- would you be willing to check on this bug using a different set of MAME thumbnails? If the devs approve this set, it will become the start of the 'official' thumbnails set for RetroArch: https://drive.google.com/drive/folders/0B4gWQ8g8Bzq3eTVqRi0zQkRvMHc

If you had the same problem with these images it would be easier for me to confirm and help you document the bug.

MopheusDG, 2016-07-31:
Hi @markwkidd - I'll test those files tonight. I don't have the complete set but I can at least check if some of the not working files are working now. Will get back to you later. Thanks.

MopheusDG, 2016-10-24:
I think I finally found the problem, it seems Lakka or RA does not like PNG files with Bit Depth of 1 or 2 bits. This is quite common on some Screenshot Sets for MAME found online to save space on games like Gun Fight, that are online black & white. I'm uploading a file with the issue.

By the way, just saving them to normal PNG's and also removing transparency (if they have) fixed the problem.

[image]

Hope this helps.

MopheusDG, 2016-10-24:
Sorry, I don't know what RPNG is, but I've tested it and if you use a PNG with 1 or 2 bits depth, RA is not showing the screenshot, if you resave the png with 8 bits or more of depth, it works.

Alcaro, 2016-10-24:
RPNG is our PNG decoder. We found libpng too bulky, we can't cram such a big thing into tiny toys like Nintendo DS.

Doesn't surprise me if it doesn't understand more rarely used features, like grayscale or transparency. I hope it at least understands paletted images.

Still, if there are grayscale images that RPNG could reasonably encounter, then grayscale should be implemented.

MopheusDG, 2016-10-24:
Oh, I got it. Was doing a search for RPNG and found their GIT repo. I don't know if the issue is worthy or not to fix, just changing bit depth and removing transparency seems to be enough to fix the problem. Of course, the file in 1 or 2 bits was only 16k and change to 160k (more or less) after changing depth, but still... easy to solve. I think it's important to let the ppl who work on thumbnails know about this, in particular for old consoles that could be having sshots with low depth.

Anyway, thanks for the support.