libretro/Lakka #424: Newer gcc

jjwatt, 2016-07-04:
I'm just getting into Lakka. The first thing I wanted to try to do was upgrade some packages in the toolchain. Are there strong technical reasons for remaining on gcc4? Since you're compiling everything, I wouldn't think it would be compatibility. My guess is that some packages aren't building right with gcc5 (or 6), and that might just mean fixing their builds. There are lots of reasons to go to a newer gcc, especially on a project like lakka: mainly, optimization improvements across platforms. The [GCC improvements] (https://gcc.gnu.org/gcc-5/changes.html) for ARM are worth noting.

Kivutar, 2016-07-04:
There are some packages that need patching to build with gcc5. For example the kernel of OdroidC1. You can find some gcc5 compatibility patches in the LibreELEC tree, or even OpenELEC. If you want to work on upgrading to gcc5, I would be happy to help you on IRC.

jjwatt, 2016-07-06:
Sounds good! I'm familiarizing myself with the build system. I'll talk to you on IRC soon.

Kivutar, 2016-07-09:
We are upgrading the kernel to 4.6.3 now. It works well. Tell us if you need help with gcc5.

Kivutar, 2016-07-29:
Can we close this now?

jjwatt, 2016-07-30:
Have you been able to compile under gcc5? Do we still need older kernels for some platforms? The last time I tried compiling with a gcc5 target on a gcc6 host, I was still having some issues. I'll have to look again, but it looked like one of the packages was using the host compiler instead of the target compiler. I'll dive in again this weekend and give an update.

Kivutar, 2016-08-25:
I've not tried gcc5. Any news on your side?

Kivutar, 2016-09-24:
No news, and we're moving to LibreELEC anyway, so I close this.

closed 2016-09-24 by Kivutar