libretro/Lakka #427: Problem Booting on 2006 iMac from USB

channemann, 2016-07-07:
Target machine:
iMac 2006
2.16 GHz Intel Core 2 Duo
2 GB RAM 667 MHz DDR2 RAM

I have tried using the [current](http://www.lakka.tv/get/macos/generic/) 32-bit build, the most recent nightly build (from July 1), and the most recent nightly EFI build (as suggested in a [previous issue thread](https://github.com/libretro/Lakka/issues/410)).

When I go to the bootloader screen (by holding alt/option after turning the machine on), only the computer's internal hard drive appears as an option.

To check whether it was an issue with the computer being too old, I also tried on a 2010 MacBook Pro, with the same result.

Any obvious debugging I should be trying? Should the USB drive be formatted a specific way prior to writing the .img file to it?

gouchi, 2016-07-09:
You can try [rEFInd](http://www.rodsbooks.com/refind) it may help but we need to know first which `https://support.apple.com/en-us/HT201518` you have.

channemann, 2016-07-10:
Looks like I've got Boot ROM Version IM51.0090.B09on the iMac.

gouchi, 2016-07-11:
So you have EFI boot, can you try to use [rEFind](http://www.rodsbooks.com/refind) and nightly EFI build ( you can use [32 bit ](http://mirror.lakka.tv/nightly/Generic.i386/Lakka-Generic.i386-devel-20160607151529-r21464-g936bd6d-efi.img.gz)or [64 bit](http://sources.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160607151111-r21464-g936bd6d-efi.img.gz) your CPU supports both).

closed 2016-08-11 by Kivutar