libretro/Lakka #433: FFMPEG support for wav files

hooby3dfx, 2016-07-16:
This is a feature request to allow the FFMPEG core to be able to play wav audio files.

(Note - let me know if I should move this to the retroarch project issues list instead.)

Currently they are not allowed to be played from the frontend (they aren't included in the supported file types). 
Additionally, when trying to play a wav file from the command line (via `retroarch -v -L /usr/lib/libretro/ffmpeg_libretro.so roms/audio_file.wav`) it results in erroneous/grinding loud audio playback.

Below is a log of what happens when attempting to play a wav file:

RetroArch [INFO] :: === Build =======================================RetroArch [INFO] :: [CPUID]: Features:
Built: Mar 19 2016
RetroArch [INFO] :: Version: 1.3.2
RetroArch [INFO] :: Git: 78af2e1
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: [CPUID]: Features:
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Overrides: no core-specific overrides found at /storage/.config/retroarch/FFmpeg/FFmpeg.cfg
RetroArch [INFO] :: Overrides: no game-specific overrides found at /storage/.config/retroarch/FFmpeg/a2002011001-e02.cfg
RetroArch [INFO] :: Environ SET_VARIABLES.
RetroArch [INFO] :: Environ GET_LOG_INTERFACE.
RetroArch [INFO] :: Remaps: core name: FFmpeg
RetroArch [INFO] :: Remaps: game name: a2002011001-e02
RetroArch [INFO] :: Remaps: remap directory: /storage/remappings
RetroArch [INFO] :: Remaps: no game-specific remap found at /storage/remappings/FFmpeg/a2002011001-e02.rmp
RetroArch [INFO] :: Remaps: no core-specific remap found at /storage/remappings/FFmpeg/FFmpeg.rmp
RetroArch [INFO] :: Content loading skipped. Implementation will load it on its own.
RetroArch [INFO] :: Environ SET_INPUT_DESCRIPTORS:
RetroArch [INFO] ::     RetroPad, User 1, Button "D-Pad Up" => "Seek +60 seconds"
RetroArch [INFO] ::     RetroPad, User 1, Button "D-Pad Down" => "Seek -60 seconds"
RetroArch [INFO] ::     RetroPad, User 1, Button "D-Pad Left" => "Seek -10 seconds"
RetroArch [INFO] ::     RetroPad, User 1, Button "D-Pad Right" => "Seek +10 seconds"
RetroArch [INFO] ::     RetroPad, User 1, Button "L" => "Cycle Audio Track"
RetroArch [INFO] ::     RetroPad, User 1, Button "R" => "Cycle Subtitle Track"
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: XRGB8888.
Input #0, wav, from 'roms/a2002011001-e02.wav':
  Duration: 00:00:54.31, bitrate: 1411 kb/s
    Stream #0:0: Audio: pcm_s16le ([1][0][0][0] / 0x0001), 44100 Hz, 2 channels, s16, 1411 kb/s
RetroArch [INFO] :: Environ GET_VARIABLE ffmpeg_temporal_interp:
RetroArch [INFO] ::     enabled
RetroArch [INFO] :: Environ GET_VARIABLE ffmpeg_color_space:
RetroArch [INFO] ::     auto
RetroArch [INFO] :: Skipping SRAM load..
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 44063.25 Hz.
[SWR @ 0x74000480] Input channel count and layout are unset
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/5xbr-retro.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bead.glsl"
RetroArch [INFO] :: [SWR @ 0x74000480] Context has not been initialized
Found shader "/usr/share/common-shaders/blinky.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/pixellate.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/quilez.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/retro-v2.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/scanline.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/sharp-bilinear.glsl"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
[SWR @ 0x74000480] Context has not been initialized
RetroArch [INFO] :: Video @ 960x720
[SWR @ 0x74000480] Context has not been initialized
RetroArch [INFO] :: Starting threaded video driver ...
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[... 122 lines cut ...]
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
[SWR @ 0x74000480] Context has not been initialized
^C[SWR @ 0x74000480] Context has not been initialized

Kivutar, 2016-07-19:
For your info, there is also an ffmpeg core built in retroarch that can play wav files, iirc.

hooby3dfx, 2016-07-19:
Thanks, it sounds like this is then a bug with the Lakka ffmpeg core/config.

Kivutar, 2016-07-22:
@twinaphex and @gouchi, should we remove the ffmpeg core in favor of the backed in one?

Is the backed in ffmpeg compatible with playlists?

gouchi, 2016-07-22:
@kivutar I think the baked in one is more updated.

I don't know if the builtin is compatible with playlists.

hooby3dfx, 2016-07-23:
BTW, I noticed that wav files actually do show up in the GUI (Select file and detect core) to be played by the ffmpeg core, so I think this should be labelled as a bug since they cannot be played correctly.

Kivutar, 2016-08-05:
Can you try these builds with the builtin ffmpeg core of RA? http://sources.lakka.tv/nightly/

hooby3dfx, 2016-08-15:
I tried the latest (20160811145614-r21565) and the issue is still there.

Kivutar, 2016-08-15:
Using the builtin player or the external core? Let try the builtin one, use Load Content Detect Core.

hooby3dfx, 2016-08-15:
I believe that is what I did.
Is the external core even present in that build?

Kivutar, 2016-08-15:
Yes.. then it also affects the builtin core.. :/

hooby3dfx, 2016-08-16:
Which means it is a retroarch issue? 
(5 mins later)
I reproduced the issue in retroarch - I can create the bug report there.

closed 2016-08-16 by hooby3dfx