libretro/Lakka #434: Lakka hangs on flower icon

leolite1, 2016-07-18:
I am using a SATELLITE PRO C850-10U laptop with latest image (Lakka-Generic.x86_64-devel-20160717204331-r21527-g4656d16) and can't get beyond the icon. I have tried many other images but with same result Ctrl+Alt+F3 gave a blank screen. I have Ubuntu 14.04 on the laptop which starts up correctly. Can Lakka work on this laptop?

Kivutar, 2016-07-18:
Hello, can you tell me the exact name of the gaphic card on this laptop
please?

gouchi, 2016-07-18:
If you can access Lakka[ command line interface ](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/)try to get a [log](http://www.lakka.tv/doc/FAQ/#how-do-i-debug-retroarch).

And if you can get the output of 
`lspci -nn | grep VGA `and `dmesg`

Thank you for your tests !

leolite1, 2016-07-19:
lspci -nn | grep VGA

> 00:02.0 VGA compatible controller [0300]: Intel Corporation 2nd Generation Core Processor Family Integrated Graphics Controller [8086:0116] (rev 09)

dmesg

> [    0.000000] Initializing cgroup subsys cpuset
> [    0.000000] Initializing cgroup subsys cpu
> [    0.000000] Initializing cgroup subsys cpuacct
> [    0.000000] Linux version 3.13.0-92-generic (buildd@lgw01-59) (gcc version 4.8.4 (Ubuntu 4.8.4-2ubuntu1~14.04.3) ) #139-Ubuntu SMP Tue Jun 28 20:42:32 UTC 2016 (Ubuntu 3.13.0-92.139-generic 3.13.11-ckt39)
> [    0.000000] KERNEL supported cpus:
> [    0.000000]   Intel GenuineIntel
> [    0.000000]   AMD AuthenticAMD
> [    0.000000]   NSC Geode by NSC
> [    0.000000]   Cyrix CyrixInstead
> [    0.000000]   Centaur CentaurHauls
> [    0.000000]   Transmeta GenuineTMx86
> [    0.000000]   Transmeta TransmetaCPU
> [    0.000000]   UMC UMC UMC UMC
> [    0.000000] e820: BIOS-provided physical RAM map:
> [    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009d7ff] usable
> [    0.000000] BIOS-e820: [mem 0x000000000009d800-0x000000000009ffff] reserved
> [    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
> [    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
> [    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
> [    0.000000] BIOS-e820: [mem 0x0000000020200000-0x000000003fffffff] usable
> [    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000401fffff] reserved
> [    0.000000] BIOS-e820: [mem 0x0000000040200000-0x00000000a79edfff] usable
> [    0.000000] BIOS-e820: [mem 0x00000000a79ee000-0x00000000a8176fff] ACPI NVS
> [    0.000000] BIOS-e820: [mem 0x00000000a8177000-0x00000000a8199fff] usable
> [    0.000000] BIOS-e820: [mem 0x00000000a819a000-0x00000000a819efff] ACPI NVS
> [    0.000000] BIOS-e820: [mem 0x00000000a819f000-0x00000000acabefff] usable
> [    0.000000] BIOS-e820: [mem 0x00000000acabf000-0x00000000acebefff] reserved
> [    0.000000] BIOS-e820: [mem 0x00000000acebf000-0x00000000acfbefff] ACPI NVS
> [    0.000000] BIOS-e820: [mem 0x00000000acfbf000-0x00000000acffefff] ACPI data
> [    0.000000] BIOS-e820: [mem 0x00000000acfff000-0x00000000acffffff] usable
> [    0.000000] BIOS-e820: [mem 0x00000000ad000000-0x00000000af9fffff] reserved
> [    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
> [    0.000000] BIOS-e820: [mem 0x00000000feb00000-0x00000000feb03fff] reserved
> [    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
> [    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
> [    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
> [    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
> [    0.000000] BIOS-e820: [mem 0x00000000ffd80000-0x00000000ffffffff] reserved
> [    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000014f5fffff] usable
> [    0.000000] NX (Execute Disable) protection: active
> [    0.000000] SMBIOS 2.7 present.
> [    0.000000] DMI: TOSHIBA Satellite C855/Portable PC, BIOS 1.60 04/20/2012
> [    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
> [    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
> [    0.000000] e820: last_pfn = 0x14f600 max_arch_pfn = 0x1000000
> [    0.000000] MTRR default type: uncachable
> [    0.000000] MTRR fixed ranges enabled:
> [    0.000000]   00000-9FFFF write-back
> [    0.000000]   A0000-BFFFF uncachable
> [    0.000000]   C0000-E7FFF write-protect
> [    0.000000]   E8000-EFFFF write-combining
> [    0.000000]   F0000-FFFFF write-protect
> [    0.000000] MTRR variable ranges enabled:
> [    0.000000]   0 base 000000000 mask F80000000 write-back
> [    0.000000]   1 base 080000000 mask FE0000000 write-back
> [    0.000000]   2 base 0A0000000 mask FF8000000 write-back
> [    0.000000]   3 base 0A8000000 mask FFC000000 write-back
> [    0.000000]   4 base 0AC000000 mask FFF000000 write-back
> [    0.000000]   5 base 0FFC00000 mask FFFC00000 write-protect
> [    0.000000]   6 base 100000000 mask F80000000 write-back
> [    0.000000]   7 base 14F600000 mask FFFE00000 uncachable
> [    0.000000]   8 base 14F800000 mask FFF800000 uncachable
> [    0.000000]   9 base 150000000 mask FF0000000 uncachable
> [    0.000000] x86 PAT enabled: cpu 0, old 0x7040600070406, new 0x7010600070106
[... 686 lines cut ...]
> [   10.925316]    mono: mono_out=0x0
> [   10.925317]    inputs:
> [   10.925319]      Mic=0x18
> [   10.925320]      Internal Mic=0x12
> [   10.925321] realtek: Enabling init ASM_ID=0xfc13 CODEC_ID=10ec0269
> [   11.029712] input: HDA Intel PCH HDMI/DP,pcm=3 as /devices/pci0000:00/0000:00:1b.0/sound/card0/input11
> [   11.029819] input: HDA Intel PCH Headphone as /devices/pci0000:00/0000:00:1b.0/sound/card0/input10
> [   11.029918] input: HDA Intel PCH Mic as /devices/pci0000:00/0000:00:1b.0/sound/card0/input9
> [   11.257568] type=1400 audit(1468925942.738:2): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/sbin/dhclient" pid=404 comm="apparmor_parser"
> [   11.257907] type=1400 audit(1468925942.738:3): apparmor="STATUS" operation="profile_load" profile="unconfined" name="/usr/lib/NetworkManager/nm-dhcp-client.action" pid=404 comm="apparmor_parser"
[... cut ...]
> [   14.928596] type=1400 audit(1468925946.406:9): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/NetworkManager/nm-dhcp-client.action" pid=715 comm="apparmor_parser"
> [   14.928601] type=1400 audit(1468925946.406:10): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/connman/scripts/dhclient-script" pid=715 comm="apparmor_parser"
> [   14.929080] type=1400 audit(1468925946.406:11): apparmor="STATUS" operation="profile_replace" profile="unconfined" name="/usr/lib/NetworkManager/nm-dhcp-client.action" pid=715 comm="apparmor_parser"
>

leolite1, 2016-07-19:
I ran retroarch command DEBUG mode and the control did not come back after a minute, so I had to interrupt it (Ctrl+Z).

> RetroArch [INFO] :: === Build =======================================
> Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX
> Built: Jul 17 2016
> RetroArch [INFO] :: Version: 1.3.4
> RetroArch [INFO] :: Git: 4656d16
> RetroArch [INFO] :: =================================================
> RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarRetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/reRetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
> RetroArch [INFO] :: Version of libretro API: 1
> RetroArch [INFO] :: Compiled against API: 1
> RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
> RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
> RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
> RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
> RetroArch [INFO] :: Video @ 960x720
> RetroArch [INFO] :: Starting threaded video driver ...
> RetroArch [INFO] :: [DRM]: Found 4 connectors.
> RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
> RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
> RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
> RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
> RetroArch [INFO] :: [DRM]: Connector 1 has 13 modes.
> RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #2.
> RetroArch [INFO] :: [DRM]: Connector 2 connected: no
> RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
> RetroArch [INFO] :: [DRM]: Connector 3 connected: no
> RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
> RetroArch [INFO] :: [DRM]: Mode 0: (1024x768) 1024 x 768, 60 Hz
> libGL: Can't open configuration file /storage/.drirc: No such file or directorRetroArch [INFO] :: Found GL context: kms
> RetroArch [INFO] :: Detecting screen resolution 1024x768.
> RetroArch [INFO] :: [EGL]: EGL version: 1.4
> libGL: Can't open configuration file /storage/.drirc: No such file or directorRetroArch [INFO] :: [EGL]: Current context: 0x7fb1880d7260.
> RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096).
> RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, RendereRetroArch [INFO] :: [GL]: Version: 3.0 Mesa 11.2.0-rc1.
> RetroArch [INFO] :: Querying GL extension: ARB_ES2_compatibility => exists
> RetroArch [INFO] :: Querying GL extension: ARB_sync => exists
> RetroArch [INFO] :: Querying GL extension: ARB_texture_float => exists
> RetroArch [INFO] :: Querying GL extension: EXT_texture_sRGB => exists
> RetroArch [INFO] :: Querying GL extension: ARB_framebuffer_sRGB => exists
> RetroArch [INFO] :: GL: Using resolution 1024x768
> RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
> RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
> RetroArch [INFO] :: Checking GLSL shader support ...
> RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
> RetroArch [INFO] :: Found GLSL vertex shader.
> RetroArch [INFO] :: Found GLSL fragment shader.
> RetroArch [INFO] :: Linking GLSL program.
> RetroArch [INFO] :: Found GLSL vertex shader.
> RetroArch [INFO] :: Found GLSL fragment shader.
> RetroArch [INFO] :: Linking GLSL program.
> RetroArch [INFO] :: [GL]: Using 4 textures.
> RetroArch [INFO] :: [GL]: Loaded 1 program(s).
> RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
> RetroArch [INFO] :: Using font rendering backend: freetype.
> RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempt$troArch [INFO] :: [udev] Adding device /dev/input/event6 as type ID_INPUT_KEY$troArch [INFO] :: [udev] Adding device /dev/input/event4 as type ID_INPUT_KEY$troArch [INFO] :: [udev] Adding device /dev/input/event5 as type ID_INPUT_MOU$troArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_MOU$troArch [INFO] :: Found joypad driver: "udev".
> RetroArch [INFO] :: ALSA: Using signed 16-bit format.
> RetroArch [INFO] :: ALSA: Period size: 1024 frames
> RetroArch [INFO] :: ALSA: Buffer size: 2048 frames
> RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
> RetroArch [INFO] :: Querying GL extension: ARB_framebuffer_object => exists
> RetroArch [INFO] :: Using font rendering backend: freetype.
> function is no-op
> RetroArch [INFO] :: Using font rendering backend: freetype.
> RetroArch [INFO] :: SRAM will not be saved.
> RetroArch [INFO] :: null: [/storage/.config/retroarch/content_history.lpl].
> RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096).
> RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096).
> function is no-op
> RetroArch [INFO] :: [KMS]: Missed 1 VBlank(s) (Frame: 6, DRM frame: 8883).

Attempting to run the same command did return control back with a segmentation fault in the log.

> ...
> RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
> RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
> RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
> RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
> RetroArch [INFO] :: [DRM]: Connector 1 has 13 modes.
> RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #2.
> RetroArch [INFO] :: [DRM]: Connector 2 connected: no
> RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
> RetroArch [INFO] :: [DRM]: Connector 3 connected: no
> RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
> RetroArch [INFO] :: [DRM]: Mode 0: (1024x768) 1024 x 768, 60 Hz
> libGL: Can't open configuration file /storage/.drirc: No such file or directo$
> RetroArch [INFO] :: Found GL context: kms
> RetroArch [INFO] :: Detecting screen resolution 1024x768.
> RetroArch [INFO] :: [EGL]: EGL version: 1.4
> libGL: Can't open configuration file /storage/.drirc: No such file or directo$
> RetroArch [INFO] :: [EGL]: Current context: 0x7fe5740d7260.
> RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096).
> Segmentation fault
>

leolite1, 2016-07-19:
Some more information

- I am connected to the secondary VGA output of the laptop; the LCD is busted which is why I was attempting to re-purpose it for Lakka. I assumed that since Ubuntu runs so Lakka would as well.

- I repeatedly pressed Escape and F1 attempting to get into BIOA just after restarting the laptop and this insted gave me the boot menu where pressing Enter gave the Install/Live option from where I installed Lakka into a different media. But rebooting with just that gave the same flower icon and same log on starting retroarch in DEBUG mode.

gouchi, 2016-07-21:
We will need to get some debug information as RetroArch crashed.

Can you try to boot to live linux distribution and [compile retroarch](http://wiki.libretro.com/index.php?title=RetroArch_Compilation) with debug symbols ?
```
git clone https://github.com/libretro/RetroArch.git
cd RetroArch
./configure --enable-kms --enable-egl && DEBUG=1 make
gdb ./retroarch
```

To launch in [kms mode](https://github.com/libretro/RetroArch/wiki/KMS-mode), you will need to launch it from a virtual console and stop xorg.

leolite1, 2016-07-24:
Okay I was able to figure out my problem. I built retroarch and then found the "-v" verbosity flag which helped me nail down the issue.

My first suspicion was on this error which turned out to be benign.
>  [ERROR] :: Failed to connect to Wayland server.

The log showed that my laptop had 4 connectors and "0" default selection.
> RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
RetroArch [INFO] :: [DRM]: Connector 1 has 13 modes.
RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #2.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1024x768) 1024 x 768, 60 Hz

The [documentation](https://raw.githubusercontent.com/Themaister/RetroArch/master/retroarch.cfg) for video_monitor_index mentioned this
> Which monitor to prefer. 0 (default) means no particular monitor is preferred, 1 and up (1 being first monitor)

So I switched it to "2" since "1" probably would be the LCD (which was busted) and this was the issue. Log now gave this and the xmb based menu showed up.

> RetroArch [INFO] :: [DRM]: Found 4 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 1 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: yes
RetroArch [INFO] :: [DRM]: Connector 1 has 13 modes.
RetroArch [INFO] :: [DRM]: Connector 1 assigned to monitor index: #2.
RetroArch [INFO] :: [DRM]: Connector 2 connected: no
RetroArch [INFO] :: [DRM]: Connector 2 has 0 modes.
RetroArch [INFO] :: [DRM]: Connector 3 connected: no
RetroArch [INFO] :: [DRM]: Connector 3 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1680x1050) 1680 x 1050, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1440x900) 1440 x 900, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1280x800) 1280 x 800, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (1152x864) 1152 x 864, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (1024x768) 1024 x 768, 70 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (640x480) 640 x 480, 67 Hz
RetroArch [INFO] :: [DRM]: Mode 11: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 12: (720x400) 720 x 400, 70 Hz

Thank you pointing

gouchi, 2016-07-25:
Congratulations ! Nice catch !

Yes you are right I should have looked more closely to video connector ;-)

Glad it works !

closed 2016-07-25 by gouchi