libretro/Lakka #436: None of the pokemon roms work on any of the emulators, for me.

wimberlyw, 2016-07-20:
I have tried all of the emulators available for Retropie, I have the BIOS and config file in several different places, just in case, but I still can not get any of the Pokemon roms to run. And I've tried several different versions of the roms, as well. I have tried loading another game, and then loading into the rom by "load another game," as I saw that was an issue, here, but it did not work for me. I have been able to get to a splash screen from whoever dumped the rom on a few emulators, but it will not load past that. I am at a loss. I really would love to get any of the games rolling.

wimberlyw, 2016-07-20:
I've actually yet to find a file of any game that actually works on any of the emulators for gba...

Kivutar, 2016-07-20:
You need the gba bios

theDripp, 2016-07-20:
Yup, check [this page](http://www.lakka.tv/doc/ROMs-and-BIOSes/) out. There's a table that mentions what bios each emulator needs, if an emulator doesn't list one then it doesn't need one.

Make sure that the bios you find match the listed MD5. As long as the MD5 matches you can rename the bios file to match the one mentioned on the table as well.

wimberlyw, 2016-07-20:
I have the bios in 3 different locations...as well as the config file. On a pi3 freshly updated everything.

cudencuden, 2016-07-21:
just answer if your bios file match the md5 from site...also answer if you planned bios in system folder

Kivutar, 2016-08-07:
You can use ssh and the md5 command to check this.

closed 2016-10-04 by Kivutar