libretro/Lakka #437: Missing some items from online updater in pc latest build

jonny555, 2016-07-20:
Hi missing items in online  updater  build 17 July 
Assets
Cores 
Database updaters are missing

fr500, 2016-07-20:
On Lakka those are meant to be handled with Lakka updates instead of the
core updater

jonny555, 2016-07-20:
Hi sorry I mean the items aren't listed at all go to online updater  the only two options are update lakka and update thumbnails rest is missing

cudencuden, 2016-07-21:
this is by design...you can only update lakka n thumbnails...

anything that is missing our needs updated will be done through updating lakka

jonny555, 2016-07-21:
I'm confused now so you don't use the online updater to get cores and update cores anymore through lakka xmb menu anymore

cudencuden, 2016-07-21:
........to get updated core...you use "update lakka" option......to update assets...you use "update lakka".....to update database...again, you use "update lakka" option....

Kivutar, 2016-07-21:
Yes, this is by design. This ensures that there is no inconsistency between the different components.

closed 2016-07-21 by Kivutar