libretro/Lakka #439: bluez.conf getting overwritten

blunckr, 2016-07-21:
According to the end of [this guide](http://www.lakka.tv/doc/Wireless-Dualshock/). To make bluez attempt to automatically connect to a device, you have a add a line to `/storage/.cache/services/bluez.conf` (`HIDD_OPTIONS="--connect xx:xx:xx:xx:xx:xx --connect xx:xx:xx:xx:xx:xx"`. I have added the line, but each time I reboot, or save my retroarch settings, this file gets overwritten with a blank file. I assume this has to do with the "Bluetooth Enable" option in the Lakka services part of the menu. I apologize if I'm just doing it wrong.

Kivutar, 2016-07-21:
You're not doing it wrong, RetroArch is indeed overwritting that file. We should check if the file exists before overriding it.

However, I don't think that line is needed to get bluez working as expected. For example, my Dualshock 3 is reconnecting automatically without that line.

blunckr, 2016-07-21:
Hmm, ok. I'm working on getting the wii remote working and this is the last step. I'll will play around with it some more and see what I figure out. I may write up a guide on working with wii remotes when I'm done, it wasn't too difficult but I've had to do a bit of googling to get it figured out.

blunckr, 2016-07-22:
I got it worked out. I was pressing the red button under the battery cover to sync, which only works  when it's first pairing ¯\_(ツ)_/¯. I didn't have to write anything into `bluez.conf`. I suppose the issue could be closed? Unless there are cases where you still might need to include something in there.

Kivutar, 2016-07-22:
Ok, glad it worked.

ruinunes, 2016-07-31:
Using a NES30 controller. The docs say we should add the MAC entry to the `bluez.conf` and I do get the file back to no contents when rebooting. I have a setup where it is the TV, that powers the raspberry pi, hence when shutting it down, pi gets shutdown too and all gets lost. Hence, I would ask to re-open the issue or in fact check if the file exists before re-writing.

Is there another option to keep the controller and the bluetooth controller to reconnect, after a reboot?

blunckr, 2016-08-01:
If you follow the guide I referenced at the beginning and go through the whole process of trusting the device, bluez should attempt to reconnect automatically, without adding anything to the file.

markwkidd, 2016-08-01:
Hi @blunckr , thank you for helping with this! You mentioned that you were willing to help document the Wii Remote. I have just cloned the Wireless Dualshock doc as a new Wii Remote doc: https://github.com/libretro/Lakka/wiki/Wii-Remote

Would you be willing to either tell me which sections of the Dualshock doc need to be removed/changed, or make those changes directly to the wiki page? 

For my part, I have a DolphinBar on the way and I'll be glad to contribute a section to the docs on how to use a Wii Remote that way. I'm don't think I'll be connecting to the remote via a standard bluetooth configuration though, so your help would be great.

blunckr, 2016-08-01:
Yes, I've been planning on doing that. I'll try to get to it as soon as I can. It shouldn't be too much different than the dualshock guide.

closed 2016-08-25 by Kivutar