libretro/Lakka #441: add ppge_atlas.zim to the default Lakka installation

markwkidd, 2016-07-23:
Of all of these various BIOS files that are required to fully utilize the various Lakka cores, one can be sourced from Lakka's own github repository: ppge_atlas.zim for the PSP. Here it is: https://github.com/lakkatv/libretro-ppsspp/blob/master/assets/ppge_atlas.zim

It's a small file, and it seems like bundling the file would be simpler for users and developers alike. Is there a reason it doesn't make sense to do so?

markwkidd, 2016-07-25:
This is a little bit more unusual than I even realized with the first post. Here are the full directions for getting the PSP core working ---- couldn't this all be built in?

"PPSSPP requires the asset files from https://github.com/libretro/libretro-ppsspp/tree/master/assets as well as the lang and flash0 folders from https://github.com/libretro/libretro-ppsspp be placed in the system\PPSSPP folder for operation. The lang folder requires updating the submodule to acquire and this can be done with 'git submodule update --init lang'.

Alternatively all of the required asset files can be downloaded or cloned with git from https://github.com/orbea/libretro-ppsspp-assets. "

https://wiki.libretro.com/index.php?title=PPSSPP

duramato, 2016-07-25:
Just leaving my experience here. At first i tought it was only the `ppge_atlas.zim` needed, becouse that was the only one mentioned in the wiki. Some games worked other didn't i then found that libretro wiki page that had that info.  And that was when it finnaly worked. I did added that info in the wiki page here in github repo, but i think it would be easier to have it come with Lakka

markwkidd, 2016-08-12:
@Kivutar will this new ppsspp issue do the trick? https://github.com/libretro/libretro-ppsspp/issues/83

Kivutar, 2016-08-12:
Mhhh no

markwkidd, 2016-08-12:
Oh no. I'm sorry -- I added a note at the bottom of that issue that there is a mistake in the post. Should I close it instead?

Kivutar, 2016-08-12:
No it's OK, you can keep this one open.

sergiobenrocha2, 2016-09-07:
About the save files and save states, it's already fixed: libretro/libretro-ppsspp/issues/75

markwkidd, 2016-09-28:
I changed the title to reflect @sergiobenrocha2 's comment about save states being standardized. Now it's just the BIOS location.