libretro/Lakka #442: dreamcast BIOS files and saved states are not handled in the standard way

markwkidd, 2016-07-24:
Per this thread the forum: http://libretro.com/forums/showthread.php?t=6405

There are two aspects of functionality that seem problematic:
1) The Reicast core requires its BIOS files to be put in a non-standard location -- within a subfolder of `roms/` named `dc`.
2) The Core is also using this `dc` for saved games, which causes a permissions issue

Both of these issues may actually be with libretro/RetroArch, but it seems like especially #2 might involve Lakka.

markwkidd, 2016-07-24:
This may be pertinent: https://github.com/libretro/reicast-emulator/issues/23

aphex-, 2016-07-24:
Setting the permissions of the **dc** folder to 777 fixes the saving issue.