libretro/Lakka #444: Lakka crash when opening playlist

duramato, 2016-07-25:
What i did: Made a playlist via lakka's scan and detect core functionality (of game boy color roms in .gbc)
What happened: When going to the console's playlist lakka crashes

Log of the crash: https://gist.github.com/duramato/3e6436d1ee43daae66b95d304ce2f0f9

markwkidd, 2016-07-25:
What happens when you load the same ROM file manually via Load Content and Detect Core?

duramato, 2016-07-25:
Opens fine and its playable.

markwkidd, 2016-07-25:
Next test, if you're willing: what if you zip up a ROM you know is working and change just its one playlist to point to the .zip?

markwkidd, 2016-07-25:
Also, could you post an entry or two from the playlist that is crashing? (6-12 lines)

duramato, 2016-07-25:
It crashes for .zip and standalone, and tbh my first experiece was with a .zip , i later tried unzipping to see if it was the .zip that was causing it. So its independat of the archieve format.
Will add a snippet of the playlist later.

duramato, 2016-07-25:
https://gist.github.com/duramato/a1c9bea1318b659b78edfff14a0800d7

markwkidd, 2016-07-26:
I am going to make my first diagnostic theory: Lakka is not DETECTing the core directly.

I honestly have a personal philosophy that user-generated playlists should specify the path to their core manually anyway.

What happens if you replace DETECT with the path to the core?

duramato, 2016-07-26:
I tried and same thing happened. 
But I would like to see an option like Scan and specify core as I think people only scan  a dir with only one kind of roms,  and I'm seeing game boy color roms beeing detected as normal game boy and vice versa  (both in a  .zip)

markwkidd, 2016-07-26:
That is the main set of troubleshooting steps I know of for your symptoms. Hopefully someone with more knowledge will join the conversation.

There has been talk about an alternative scanning method. I tend to agree with your approach for what it's worth: https://github.com/libretro/RetroArch/issues/2033#issuecomment-234686250

duramato, 2016-07-28:
Full playlist that is crashing https://gist.github.com/duramato/252558204911f6fb7be78ea02079c44a

Kivutar, 2016-07-29:
Are you using the no-intro roms? If not, then you should.

duramato, 2016-07-29:
I don't know if they are, is there an easy way to know? To be noted that they do open fine in the emulator is just the playlist that causes the crash. And a collection of normal game boy roms i have from the same source doesn't crash when navigating the playlist.

Kivutar, 2016-07-29:
Let try to delete your playlist, and check if you get the same problem with less roms.

Kivutar, 2016-07-29:
I suspect a special filename

duramato, 2016-07-29:
i tried with a handfull of roms like 10 and the issue wasn't present (i have ~1000+ roms), but i can try again. About a special filename i don't know of an easy way to find it they are 1000 roms

Kivutar, 2016-07-29:
Yeah. It's going to be difficult to know the real cause of the bug without finding the file that triggers it. So it all depends on you.

markwkidd, 2016-08-05:
@duramato I quickly browsed through your full playlist file. There are definitely some filenames & corresponding playlist entries with characters that are 'forbidden' by the RetroArch filename convention in your playlist.

Could I ask if you are you using a ROM set that is in public circulation, or one that you have compiled and validated yourself? 

A similar issue came up with my MAME 2003 playlist and thumbnail set. I dealt with it via some scripting. Here is a little bit of a AutoHotKey (Windows) script to sanitize filenames for use with RetroArch. Forbidden characters are replaced with an underscore ("_").

Something like this might be adaptable to your situation if you are familiar with any script languages that can bulk rename your ROMs:

    fancyname := filename   ;### the file is not matched in the dat file, use the filename instead
    ;### Replace characters unsafe for cross-platform filenames with underscore, 
    ;### per RetroArch thumbnail/playlist convention
    fancyname := StrReplace(fancyname, "&apos;", "'")
    fancyname := StrReplace(fancyname, "&amp;", "_")
    fancyname := StrReplace(fancyname, "&", "_")
    fancyname := StrReplace(fancyname, "\", "_")
    fancyname := StrReplace(fancyname, "/", "_")
    fancyname := StrReplace(fancyname, "?", "_")
    fancyname := StrReplace(fancyname, ":", "_")
    fancyname := StrReplace(fancyname, "<", "_")
    fancyname := StrReplace(fancyname, ">", "_")
    fancyname := StrReplace(fancyname, ":", "_")
    fancyname := StrReplace(fancyname, "*", "_")
    fancyname := StrReplace(fancyname, "|", "_")

duramato, 2016-08-16:
@markwkidd i don't know how to use AutoHotKey, do you know the full list of forbidden char? So i can something like what you made.
The ROMs are in semi public circulation.

cudencuden, 2016-08-16:
i believe his snip of code above listed all of the forbidden characters.

Kivutar, 2016-08-25:
It's certainly not & or < or >

Kivutar, 2016-08-25:
I downloaded your playlist and loaded it in my Lakka without any crash. The description says

> When going to the console's playlist lakka crashes

Or should it be "When I'm launching a game from the console's playlist lakka crashes" ?

duramato, 2016-08-26:
It is correct, when going to the play list. 
Maybe works for you becouse you don't have the roms? Idk

Kivutar, 2016-08-26:
Can you try to delete your roms then? I doubt it would make a change.