libretro/Lakka #445: Help lakka caught in flower fabric

Dallignaantonio, 2016-07-25:
I am new to this subject and I can not install lakka tried these two (Lakka - Generic.x86_64 -devel- 20160710153555 - r21519 - gd0ce46d and Lakka-Generic.i386-devel-20160607151328-r21464-g936bd6d). After all installation it hangs in the flower display. so something happens to turn off the PC. appears following error message failed unmounting / flash.
Please help me

TheDcoder, 2016-07-26:
It might be an incompatibility with your PC or your PC's BIOS. We need more information, what OS do you use?

Dallignaantonio, 2016-07-27:
I had installed windows xp there formated hd to put lakka

TheDcoder, 2016-07-27:
ok, did you try running lakka in live mode?

Dallignaantonio, 2016-07-28:
the same thing happens it hangs in Flower fabric

TheDcoder, 2016-07-28:
Hmm... Do you have Legacy mode enabled in your BIOS?

TheDcoder, 2016-07-31:
It differs from computer to computer, a little googling might help! also, check your computer/motherboard manufacturer's documentation about your BIOS, it might contains useful information.

gouchi, 2016-07-31:
Try to connect to lakka command line interface to get some [log](http://www.lakka.tv/doc/FAQ/#how-do-i-debug-retroarch)  provide the output of 
`lspci -nn | grep VGA`

gouchi, 2016-08-21:
Do you reproduce the issue with [latest nightly ](http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160811142524-r21565-gc79651a.img.gz)?

Kivutar, 2016-10-04:
No news, we close

closed 2016-10-04 by Kivutar