libretro/Lakka #446: Cannot move mouse pointer in RetroArch

TheDcoder, 2016-07-26:
Hello! I have enabled the mouse pointer setting in RetroArch and I can see the cursor at the top left side of the screen but I cannot move it... I can confirm that my mouse is not a fault be cause I can use my mouse while playing NDS games in the DeSmuMe core.

markwkidd, 2016-07-26:
I can confirm the same symptoms on a Thinkpad x61 running the most recent build via Live USB with the built-in Thinkpad trackpoint and a USB-attached Dell mouse.

TheDcoder, 2016-07-27:
Thanks for confirming @markwkidd :+1:

Kivutar, 2016-10-11:
It also affects me.

TheDcoder, 2016-10-11:
Additional information: I have only tried it on PC, has anyone else tried it on other hardware like RPi, Odroid etc?

luca-saggese, 2016-12-31:
same problem on raspberrry pi 2

TheDcoder, 2016-12-31:
Thanks for confirming