libretro/Lakka #447: Dreamcast core crash (Reicast)

r6turboextreme, 2016-07-26:
The Dreamcast core crashes upon loading a game. I used a .cdi file from the game Silver. Lakka does crash, it needs a power cycle. Any idea how to solve this?
Thanks

markwkidd, 2016-07-31:
Is there any way you can get a system log of the crash to post here?

There are some partial instructions on how to access the log here: https://github.com/libretro/Lakka/wiki/Troubleshooting-Lakka

Yqnn, 2016-08-06:
I have a same issue on a Core i5-4258U, with 2 different cdi files.
`RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 384 frames
RetroArch [INFO] :: ALSA: Buffer size: 1536 frames
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: null: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: [GL]: VSync => on
context_reset.
Linux paging: 00001000 00001000 00000FFF
RetroArch [libretro INFO] :: Verify Failed  : ftruncate(fd,RAM_SIZE + VRAM_SIZE +ARAM_SIZE)==0
 in _nvmem_alloc_mem -> core/hw/mem/_vmem.cpp : 591 
DEBUGBREAK!`

Yqnn, 2016-08-06:
It was an issue with the bios, I fixed it by moving the bios file to system/dc, as described here : https://github.com/libretro/Lakka/wiki/ROMs-and-BIOSes .

markwkidd, 2016-08-07:
It sounds like your situation is related to this one: https://github.com/libretro/Lakka/issues/442

If everything is working now, you might want to close this issue though.

Kivutar, 2016-08-07:
So it's just a permission problem on the dc folder?

markwkidd, 2016-08-07:
I know I'm butting in here, but I've seen a couple of people struggle with this on the forum too. 
This core seems to be tripping users up because instead of putting the Dreamcast BIOS files directly in the system folder, they need to create a subfolder and put the two BIOS files there instead.

This is similar to the PSP core issue ( https://github.com/libretro/Lakka/issues/441) , but I don't think this one also involves a permissions problem with the new folder unless @Yqnn says differently.

Kivutar, 2016-08-11:
Can I close this now?

Yqnn, 2016-08-12:
From my point of view, yes.
But I don't know if @r6turboextreme issue was the same as mine.

r6turboextreme, 2016-08-13:
Yes. This issue can be closed - as I have no time right now to test it, but some possible solutions are shown. I'll test it in three weeks. If this does not work yet, I'll reopen this. Thanks!

closed 2016-08-13 by r6turboextreme