libretro/Lakka #450: PC Download gives me a Linux file?

Newyears1978, 2016-07-30:
When I download Win7 64bit version I get an image that when opening has more img files and says Linux filesystem...?

I may be missing something?

Kivutar, 2016-07-30:
Yes,

We are asking you your current OS to give you installation instructions
adapted to your OS.

Lakka is a linux based OS.

If you want a program like lakka that you can setup on windows, so download
RetroArch instead.

Newyears1978, 2016-07-30:
I was confused totally. I didn't see the next steps of the installation I am sorry.
I thought Lakka was like a Retroarch branch with more features but now I see it's a way to install Retroarch as an OS.
Awesome!  Sorry about my stupid post

   

Jean-André Santoni

Kivutar, 2016-07-30:
No problem!!

closed 2016-07-30 by Kivutar