libretro/Lakka #451: Lakka on Odroid XU3

ChromoX, 2016-08-01:
Latest nightly `Lakka-OdroidXU3.arm-devel-20160715155409-r21527-g4656d16` release won't boot. The HDMI cable is active, but displays a black screen. I've attempted and failed to login to SSH on XU3. It makes me believe that the boot is not completing. I've also tried the suggestions  on the Lakka FAQ and have modified the boot.ini file with several independent attempts at different  video setenv calls. Nothing has worked thus far. Anyone got any ideas?

Kivutar, 2016-08-01:
I'm running this release myself on the XU3 and it's working fine for me.

ChromoX, 2016-08-01:
Does it make a difference that it's a XU3-Lite?

Kivutar, 2016-08-01:
I don't think so. Can you try reflashing? There are often flashing bugs.

ChromoX, 2016-08-01:
I've reflashed at least 10 times with different nightlies. The commands I've tried are the following:

`sudo dd if=Lakka_image.img of=/dev/rdisk2 bs=1m`
 and
`sudo dd if=Lakka_image.img of=/dev/rdisk2`

Kivutar, 2016-08-01:
Ok if you tested it that much, then it may be an incompatiblity with the Lite :(
You can join our IRC chan and ask mdrjr, he's working at hardkernel and can inform you.

ChromoX, 2016-08-02:
@mdrjr I tried to message you on IRC without any response. Do you have any ideas for how I can get Lakka to boot on a XU3-Lite?

mdrjr, 2016-08-02:
If you can't SSH to the board this is not a HDMI issue.
What's the LED Status? Do you have the UART kit to capture the actual error?

ChromoX, 2016-08-02:
Yeah I definitely can't find the XU3-Lite on my network even though it's plugged in via ethernet. I took a quick video of trying to boot the latest Lakka image on an SDCard. It's located at [XU3-Lite Bootup](https://vid.me/zs6G). At the end of the video the lights stop blinking and that's how it will stay until I unplug. During bootup the display thinks that the HDMI link is active, but the screen is black. I have two seperate XU3-Lites that I have tested this on. Both give the same result. I don't have a UART kit.

Kivutar, 2016-08-02:
You haven't forgot to unzip the image before flashing it, right?

ChromoX, 2016-08-02:
Definitely unzipped it. Flashed this exact file: `Lakka-OdroidXU3.arm-devel-20160715155409-r21527-g4656d16.img`, ~306MB

ChromoX, 2016-08-02:
Ordered a UART kit. I'll post logs once I receive and hook it up. If you happen to have any other ideas please let me know.

mdrjr, 2016-08-02:
Ok, I saw your video.. Bootloader runs, kernel runs and stop.

Try our Official Ubuntu image and let me know if it works.

ChromoX, 2016-08-03:
I was able to boot the Ubuntu server image with no problems.

mdrjr, 2016-08-03:
Ok, I hate to go this way but.. if Ubuntu Server works just fine and Lakka doesnt... 
Its something on the Lakka image :(

Kivutar, 2016-08-03:
Maybe the u-boot I ship is out of date and was not yet compatible with the Lite version?

Kivutar, 2016-08-03:
Can you try this new image? I tested it on my XU3, and there is an up-to-date u-boot and kernel http://static.kivutar.me/Lakka-OdroidXU3.arm-devel-20160803184825-r21549-gda7fd37.img.gz

ChromoX, 2016-08-03:
Thanks for the idea, but sadly no fix. Bootloader runs, kernel runs and then a stop. My UART kit shipped today. So in a couple of days I'll have an actual error message for you guys.

LA-Dude, 2016-08-05:
I just flashed the newest nightly 20160805 on my XU3-Lite and it boots up perfectly to the Lakka menus. Just wanted to let you all know since it seems there may be something else going on with ChromoX's device.

ChromoX, 2016-08-05:
So I received my UART kit today and tried to boot again. [Log](https://gist.github.com/ChromoX/53b7477b3c649d0a663d82871d02d9c9) I noticed something about the SD card. I have a feeling it's something with the sdcard. You guys got any ideas?

Kivutar, 2016-08-06:
I will try to flash on SD card tomorrow.

ChromoX, 2016-08-07:
After studying the log for a little bit I'm pretty sure that the cause is from this error:

`Starting kernel ...

[   12.306084] [c0] Internal error: Oops: 5 [#1] PREEMPT SMP ARM`

I've searched extensively and can't really find much. I assume you guys have not seen this before?

As for more information about this particular case:
I recently bought these from ameriDroid, and got "Recertified" Odroid-XU3-Lites. Both boards have booted these [Ubuntu Images](http://odroid.in/ubuntu_14.04lts/).

Any help would be greatly appreciated.

Kivutar, 2016-08-07:
Please post the full log?

@ChromoX have you tried eMMC?

I will try on SD.

ChromoX, 2016-08-07:
Full Log: https://gist.github.com/ChromoX/53b7477b3c649d0a663d82871d02d9c9

I don't have eMMC. I was really trying to get the SD Card to work.

ChromoX, 2016-08-11:
I gave up. Bought a XU4 and it booted right up. 

Thanks for the help!

closed 2016-08-11 by ChromoX