libretro/Lakka #452: Ghouls 'n' Ghosts / Daimakaimura

mpdt, 2016-08-01:
As of right now, this game loads properly and the attract sequence plays fine, however it crashes/resets back to desktop whenever I attempt to "insert a coin" using either the Xbox 360 wired pad or using the 5 key on a keyboard.

I have tried a 0.78, 0.139 and 0.37b5 rom, all of them have the same issue.  I have not altered any of the default settings for this game or MAME in general.

Cheers.

Kivutar, 2016-08-01:
What exact core are you using?

Rocklima2, 2016-08-03:
Use fba-next working perfect i have this rom

mpdt, 2016-08-03:
Hi Kivutar, sorry for the delayed response - I was using mame 2003, the default mame driver in the latest download version.

mpdt, 2016-08-03:
Thank you for the tip Rocklima2, I shall give that a try! :-)

Rocklima2, 2016-08-03:
Capcom and neo geo games fba-next is best option

Kivutar, 2016-08-07:
Can we close this report now?

closed 2016-08-11 by Kivutar