libretro/Lakka #453: Running Lakka on an Intel Computestick

piedoom, 2016-08-01:
I was hoping to run Lakka on my Intel Computestick (with the 64bit version as I mainly want n64 ROMs).  However, I'm running into some problems - neither the 64 bit version or 32 bit version of Lakka is recognized in the uefi.  

I did have limited succcess with the `efi` image [here](http://sources.lakka.tv/nightly/Generic.i386/Lakka-Generic.i386-devel-20160607151529-r21464-g936bd6d-efi.img.gz), although that's 32bit to my understanding.  I was able to see and boot from the USB, but then a randomly colored screen appeared with no options to install/do anything, really - just a blank screen with a random color each time I booted.

I don't understand very much about bootloaders/uefi, but to my understanding, the Intel Computestick has a 64 bit processor, but a 32 bit UEFI.  What is the best way I could get Lakka running - I've tried several different releases - is there a way I can build Lakka with the settings I need?

Thanks.

Kivutar, 2016-08-01:
Use the -efi sufixed images from this repo http://sources.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160731121452-r21547-g170e484-efi.img.gz

piedoom, 2016-08-01:
Drats, I can't believe I missed that.  Booted fine, I'm installing now.  Thanks for the response.

closed 2016-08-01 by piedoom