libretro/Lakka #454: How to rotate screen via SSH

piedoom, 2016-08-01:
Hi, 
My current setup has my monitor upside-down.  Unfortunately, I don't have any settings on my device BIOS or monitor that can flip the picture.  Normally, I would correct this problem with xrandr but it would seem Lakka does not use X(?).  All the forum posts I have seen about the topic are outdated an no longer work.  
I see there is a screen rotation setting on the menu, however that seems to only apply to games, and not the entire system.

Kivutar, 2016-08-02:
There may be a cmdline argument that you can pass to the kernel to tell drm/kms to rotate the display.

duduke, 2016-08-08:
What is your GPU/iGPU model?

piedoom, 2016-08-08:
I believe it has an Atom processor w/ intel graphics - [I'm using a compute stick.](http://www.newegg.com/Product/Product.aspx?Item=N82E16883800005)

duduke, 2016-08-10:
I think it can only be done at boot using drm kms parameters. As LAKKA is no longer using Xorg, the use of xrandr (a command line tool) is not possible. 
At least this is what I came up with. Sorry.

gouchi, 2016-08-11:
We will have to expose [kms properties](https://01.org/linuxgraphics/gfx-docs/drm/drm-kms-properties.html)  to RetroArch. For now,  you can only test it with [modetest](https://wiki.linaro.org/xinliang/libdrm/modetest).

piedoom, 2016-08-11:
Thank you all for your info!  Should I leave this issue open until such a feature is implemented in Lakka, or should it be closed thanks to the guides posted?

gouchi, 2016-08-12:
Modetest is just a testing tool.  We should open a feature request on RetroArch issue tracker and link to this issue.