libretro/Lakka #455: PlayStation games not scanning [REDUMP verified]

Jordon-H, 2016-08-02:
All my PlayStation games are in .bin/.cue format and verified REDUMP as per size, cue, MD5 and SHA-1  yet a large portion of my games aren't scanning at all.

These files are being served from a  NAS, but I get the same issue if put on the internal MicroSD as well. I tried rescanning with the playlist intact, as well as deleting the playlist, and rescanning and produce the same issue.

All other console games scan 100% when using the No-intro database i.e.: NES, SNES, Genesis, final Burn Arcade etc...

cudencuden, 2016-08-02:
http://www.lakka.tv/doc/ROMs-and-BIOSes/

Jordon-H, 2016-08-02:
What exactly are you pointing to here? I have read the documentation in regards to this and it does not solve this issue.

cudencuden, 2016-08-02:
there is no defined dat file for PlayStation games. read the first paragraph of ROM scanning and playlist generation

Jordon-H, 2016-08-03:
Thank you for clarifying what you meant. I wonder why that is - I honestly thought the "ROM scanning and playlist generation" doc was outdated.

I will close this and open a new issue to add a proper dat for PlayStation.

closed 2016-08-03 by Jordon-H

cudencuden, 2016-08-03:
you can check scripting section in http://www.lakka.tv/doc/Playlists/ to generate your own playlist...that should be okay as a work around for now