libretro/Lakka #456: Add PlayStation database [REDUMP]

Jordon-H, 2016-08-03:
As was pointed out tot me here: http://www.lakka.tv/doc/ROMs-and-BIOSes/

There is no specified database for PlayStation at this time. This causes a lot of scanning issues, and as **Redump** is essentially the **no-Intro** of the disc world, I would like to request this be used at the database for PlayStation.

I am not sure how to mark this as an enhancement, as it would fit there as well, but is sorely needed for PlayStation use.

Kivutar, 2016-08-03:
Our current scanning for PSX relies on serial numbers, because checking the sum of large files takes too much time. So we need a dat that contains the serials.

markwkidd, 2016-08-03:
I'm confused -- in the original forum post this issue is described as: "They all scan 100% using RetroArch on both Windows and OSX" but not Lakka. http://libretro.com/forums/showthread.php?t=6510

Does RetroArch have a different scanner than Lakka?

Kivutar, 2016-08-03:
Lakka lags one version behind RetroArch sometimes. You can try the last experimental build that ships RetroArch 1.3.6 too see if you get better results: http://sources.lakka.tv/nightly/

cudencuden, 2016-08-03:
what's a serial for psx???

Kivutar, 2016-08-03:
numbers like this https://github.com/libretro/libretro-database/blob/master/dat/Sony%20-%20PlayStation.dat#L8

cudencuden, 2016-08-03:
thanks...so we should rename our files to have serial number in the file name?

Jordon-H, 2016-08-03:
I realize it can take a long time to scan, but it would probably fix the issue that is present with the scanner right now. I know when I scan my FBAlpha games (about 4000 files) it takes a long time, but I just run it at night before bed. No harm done!

Kivutar, 2016-08-03:
You can't compare the size of the FBA roms to the size of the PSX roms.
However, there are other strategies, if people use only redump files, we can check the CRC of the cue, which is very small.
But it would cause another bug, on PSP this time.

Jordon-H, 2016-08-03:
Fair enough. Most people probably don't have 4000 PlayStation games though ;) I'm just saying if it took a few hours to scan a typical library - it would be worth it (IMO) versus renaming files or having to create a playlist from scratch or use scripts that are _not_ user friendly for the average user.

I would very much like the red ump feature added for PlayStation scanning as like I said earlier - it's basically the equivalent of no-intro - but for discs.

Kivutar, 2016-08-03:
If you're using PC, I can understand what you mean. But on weak hardware like a raspberry pi, scanning is slowed down by a factor of 10.

Kivutar, 2016-08-04:
@cudencuden, no, the serial number is hidden somewhere in the .bin or .iso or .img.
RetroArch parses the data in this file and gets the serial, then compare it to the database.

In 1.3.6, they improved that code a lot.
Current release of Lakka is still using 1.3.5.

Kivutar, 2016-08-05:
I uploaded builds with 1.3.6 here http://sources.lakka.tv/nightly/

Kivutar, 2016-08-18:
@RobLoach is adding REDUMP support to libretro-database. Once it's complete, we'll switch to it.

closed 2016-08-18 by Kivutar

RobLoach, 2016-08-18:
@Kivutar [libretro-dats](https://github.com/robloach/libretro-dats) builds Sony - PlayStation.dat from Redump, TOSEC, and Trurip just fine. When running libretro-build-database.sh, it errors out since there is metadata that has serial information, but no CRC associated with it.

We could either remove all the Sony PlayStation metadata and bring in the new DAT, or add the CRCs to the metadata. Thoughts?

RobLoach, 2016-08-18:
[Sony - PlayStation.dat.txt](https://github.com/libretro/Lakka/files/424959/Sony.-.PlayStation.dat.txt)

Kivutar, 2016-08-18:
I'd say we remove the metadata, and we base all on the new dat.