libretro/Lakka #457: Immediate reboot when attempting to start kernel from boot menu on Chromebox

trapexit, 2016-08-04:
Using the latest SeaBIOS. Am able to boot Windows and Ubuntu installs via the same USB stick.

HP Chromebox w/ Core i7-4600U & Intel HD Graphics 4400 via HDMI

OpenELEC 6.0.3 also exhibits this issue.

gouchi, 2016-08-05:
Can you try to boot with one these [parameters] (http://wiki.openelec.tv/index.php/Boot_Arguments) 
Just press Tab in the boot menu to edit command line. 

Try first with 'textmode debugging' . 

After try with break=boot_step. 

[Source forum] (http://libretro.com/forums/showthread.php?t=4836&p=44184)

trapexit, 2016-08-05:
No difference with either. Tried appending to the extra arguments and replacing them. On the EFI image and legacy.

gouchi, 2016-08-05:
We need to get some log,  information about your issue. 

Did you notice any error message when you booted with debug options?

trapexit, 2016-08-05:
I understand that but there is no information to provide. As I mentioned the arguments made no difference. None. It goes to a black screen and reboots almost immediately after hitting enter. If I remove quiet you see it saying it's loading the kernel and then it reboots.

gouchi, 2016-08-05:
Did you try break options ? For example 

break=load_modules ?

trapexit, 2016-08-05:
Aren't those break points for after the kernel is loaded? In the [init](https://github.com/OpenELEC/OpenELEC.tv/blob/master/packages/sysutils/busybox/scripts/init#L701) script?

This doesn't get that far. The kernel doesn't start. Or if it does it's leading to a reboot before *anything* is printed on screen.

Kivutar, 2016-08-05:
Have you tried the EFI images?

trapexit, 2016-08-05:
As I mentioned I tried both. They both almost immediately reboot after the kernel is supposed to be loaded. At most you see the syslinux loading kernel text and then black and then reboot.

trapexit, 2016-08-06:
LibreELEC 7.0.2 also exhibits this behavior.

gouchi, 2016-08-06:
Can you try to append boot option **acpi=off** it seems old kernel required this option to boot fine.  Even if we are using kernel 4.6,  it may help.

trapexit, 2016-08-06:
No go. 

Could it be syslinux? I tried RemixOS which uses v4.03 and it boots fine. Appears the latest nightly of Lakka is using 6.03. Tried booting a different kernel (the RemixOS one, and ubuntu) and it behaved the same way.

trapexit, 2016-08-10:
https://mrchromebox.tech

"and Syslinux 6.0.x doesn't boot at all (use 5.10 or 6.04-pre1)."

trapexit, 2016-08-10:
That was it. The image boots fine with syslinux 6.04-pre1.

gouchi, 2016-08-11:
Nice found,  congratulations! 

So we will have to update to version 6.04 when available to support chromebox.

Kivutar, 2016-08-25:
6.04 is not yet stable, from what I see here http://repo.or.cz/syslinux.git

Is it worth switching now?

trapexit, 2016-08-30:
It seems stable to me.