libretro/Lakka #458: Display IP Address in XMB

duduke, 2016-08-06:
When using a system with RetroPad remote and no physical controllers, it would be useful to see the IP address of the LAKKA machine.

Kivutar, 2016-08-07:
I'm not familiar yet with the remote retropad. How do you start it on both side?

duduke, 2016-08-07:
It's a part of RetroArch 1.3.6 (at least on Windows/MAC). When starting RetroArch, select start retro pad remote core from the main menu. 
I wish there was a specific retropad client without requiring the full RetroArch.

Kivutar, 2016-08-07:
That "Start RetroPad" have to be done on the pad side right? How to start listening to it on the Console side?

duduke, 2016-08-07:
Start RetroPad is on the client side, on the server side there used to be a settings option under network, but now it's in the config:

`network_remote_enable_user_p1 = "true"
network_remote_enable_user_p2 = "false"
network_remote_enable_user_p3 = "false"
network_remote_enable_user_p4 = "false"
network_remote_enable_user_p5 = "false"
network_remote_enable_user_p6 = "false"
network_remote_enable_user_p7 = "false"
network_remote_enable_user_p8 = "false"
network_remote_enable_user_p9 = "false"
network_remote_enable_user_p10 = "false"
network_remote_enable_user_p11 = "false"
network_remote_enable_user_p12 = "false"
network_remote_enable_user_p13 = "false"
network_remote_enable_user_p14 = "false"
network_remote_enable_user_p15 = "false"
network_remote_enable_user_p16 = "false"
network_remote_enable = "false"
network_remote_base_port = "55400"
`