libretro/Lakka #463: WeTek Play error on boot since specific nightly build

lookshe, 2016-08-13:
Regarding to the discussion in #282 I open a new issue.

What I have done so far:

Compiled and tested image till commit 4b7146df62fe7308bf3866b39e53818a780f9f38 without changes.

Following commit dde401dd3c29ea2be12e44e0f836e127bbabd2cb does not compile, so I merged necessary changes from 6429dc7225fcb16bd3471ae1fe29c94e16346629 to make it compile again.
> modified:       projects/WeTek_Play/linux/linux.arm.conf
> deleted:       projects/WeTek_Play/patches/linux/000-svn_rev.patch
> deleted:       projects/WeTek_Play/patches/linux/010-revert_default_dvb_core_code.patch
> deleted:       projects/WeTek_Play/patches/linux/070-remove-amvideocap-spam.patch
> deleted:       projects/WeTek_Play/patches/linux/100-fix-23hzpixelclock.patch
> deleted:       projects/WeTek_Play/patches/linux/120-enable_bluetooth.patch

After flashing the image, the same error occures like described in #282 
> sh: can't access tty; job control turned off

Kivutar, 2016-08-13:
Thanks,

Do you remember why the WeTek Play image doesn't compile after dde401d ? What was the broken package and the error message at that point?

Can you try to add 02744500d4d0b52498fafb88283f0e3b3da4b865 too?

What bugs me is that `sh: can't access tty; job control turned off` usually happen when the system fails to mount the second partition. So it could be something in the kernel config that have to do with filesystems.

Can you try to get a bootlog?

lookshe, 2016-08-13:
IIRC there were problems while executing the deleted patch-files. That's the reason why I want to try to get the patches working.

I will add 02744500d4d0b52498fafb88283f0e3b3da4b865 afterwards.

Are there any other posibilities to get a boot log? Because I cannot find my RS-232 cable...

Kivutar, 2016-08-13:
We should ask on wetek forums if there are differences between the wetek play hardware.

And also, you could try the latest OpenELEC, the kernel config and the patches are directly copied from there. Maybe they fixed it since then.

lookshe, 2016-08-14:
I compiled https://github.com/OpenELEC/OpenELEC.tv/commit/79e5aec6c97cbd40126cf17c933ea2f621ab2e80 which firstly uses the same kernel that Lakka uses and got the same error while booting. The patches from OpenELEC which sligthly differs did not seem to make a difference.
As you said, I am going to compile latest OpenELEC and afterwards I will take a look at the wetek forums and ask them for help.

lookshe, 2016-08-15:
Latest OpenELEC give me the same error.

I opened a new discussion in wetek forum: http://www.wetekforums.com/v/index.php?p=/discussion/28166/

Will try to get a boot log as soon as the USB-to-RS232 adapter arrives.

Kivutar, 2016-08-15:
Thank you. That's somehow lucky that the bug also happens on OE. Since they have official WeTek support, they may know what is the cause.

Kivutar, 2016-08-18:
Any news from this?

lookshe, 2016-08-18:
They said the problem is my old NAND installation, which is not able to boot newer kernel from SD-card. But this confuses me, because it happens since a specific kernel and not in generell with a newer kernel. I will try to confirm it with a boot log, as soon as possible. I think there were also some changes in the kernel config or somewhere else that led to this behaviour and not only because it is a newer kernel.

Kivutar, 2016-08-25:
Is it better with the new NAND bootloader? The bootloader can be incompatible with some kernels?

lookshe, 2016-08-31:
I didn't received my USB-to-RS232 adapter yet, so I was not able to get a boot log, which should tell me, that the kernel is incompatible, which still sounds a bit strange to me.
Even though I do not want to update my NAND installation, I will try to do this after I received the adapter and made the boot log.
Sorry that it takes so long, but I am also annoyed, that the adapter did not arrive after two weeks...

lookshe, 2016-09-07:
Finally I was able to make a boot log, after I had to recover my WeTek. I installed OpenELEC 5.0.8 on NAND to get the same error again. After taking the boot log I installed 6.0.3 because I had to reinstall everything, so I used a newer version. With this I was able to boot new OpenELEC images from SD card, but still have problems booting Lakka. I will investigate that in the next days, but first the boot log of NAND 5.0.8 trying to boot latest Lakka from SD card: http://www.thehappy.de/~lookshe/pub/wetek_lakka.log

Kivutar, 2016-09-07:
Can you try this new image based on LibreELEC? http://static.kivutar.me/Lakka-WeTek_Play.arm-8.0-devel-20160907223943-r23730-g012092c.img.gz

lookshe, 2016-09-10:
This image is working booting on the first try :)
But I have some problems with the video output:
[image]

Raxone, 2016-09-18:
Same problem like lookshe say.Problem with video output.

Kivutar, 2016-09-18:
Can you boot on Android and change the resolution to 1080p?

Raxone, 2016-09-18:
u-boot is last android dualboot 1.1.7 but i not use android i got enigma2
on nand.
version Lakka-WeTek_Play.arm-devel-20160109152500-r21276-g4dadcab.img
workin with old u-boot without dualboot(change boot option with remote) and
openelec or enigma2 on nand but with last u-boot dualboot not turn on hdmi
must change source on tv and than work.

Raxone, 2016-09-19:
If install android and set 1080p dualboot not work don't boot from sdcard no matters what is on sdcard enigma2,lakka or openelec.If erase nand and try boot lakka same problem with video output.
I can easy turn on ssh but i don't know what i need change to try fix problem.

Raxone, 2016-09-19:
I extract from Lakka-WeTek_Play.arm-devel-20160109152500-r21276-g4dadcab.img from kernel initramfs script "platform_init" and replace it in last lakka kernel build initramfs and rebuild kernel now work with all version.
I use enigma2/Openatv on nand,last u-boot (dualboot) and lakka on sdcard.

Kivutar, 2016-10-17:
So our current platform_init script is broken?

Raxone, 2016-10-17:
For me only change platform_init  fix problem. 
I try 2 days all possible solution but nothing fix problem.
Everething else work ok.

Kivutar, 2016-10-17:
OK, I'm trying the platform_init from LibreELEC.

Kivutar, 2016-10-17:
Could you try this please? http://static.kivutar.me/Lakka-WeTek_Play.arm-devel-20161017145155-r21617-g0d2198b.img.gz

Kivutar, 2016-10-19:
@lookshe could you give it a try please?

Raxone, 2016-10-19:
O sorry i try and got same error.

lookshe, 2016-11-15:
Sorry for the late response. I get the same result like before :(

Kivutar, 2017-03-05:
What about the new LE based images? http://le.builds.lakka.tv/

Raxone, 2017-03-05:
Same video output error again on wetekplay 1.

Kivutar, 2017-03-05:
So LibreELEC is incompatible with wetek play 1? Can you try LibreELEC 8.0 on it please?

Raxone, 2017-03-06:
I try last image
Lakka-WeTek_Play.arm-8.0-devel-20170305041324-r25834-g76ebcbd.img and error
is same.

Kivutar, 2017-03-06:
Thanks for your patience. If you have time, can you test this one please? http://releases.libreelec.tv/LibreELEC-WeTek_Play.arm-8.0.0.img.gz

Raxone, 2017-03-06:
This image work.

Kivutar, 2017-03-06:
@Ntemis, any idea of why it LE 8.0 works and not Lakka-LE 8.0? The codebase is almost 100% the same.

Ntemis, 2017-03-06:
Yes lakka needs to force a resolution from platform init or setenv to it from bootloader

Kivutar, 2017-03-06:
The same kind of things codesnake pushed on Hub and Play 2?

Ntemis, 2017-03-06:
Yes but better set at 720p vs 1080p

Kivutar, 2017-03-06:
Why?

NuisMons, 2017-03-29:
problem still present, only `fbset -fb /dev/fb0 -g 1280 720 1280 1440 32` in ssh show normal video
can`t set fullhd
my wetekplay:
WeTekPlay Openelec edition with
libreelec 8.0.1 on nand

Ntemis, 2017-03-30:
Can you move this issue to lakka libreelec please?