libretro/Lakka #464: corrections to "Arcade" doc page

markwkidd, 2016-08-14:
The new arcade doc page incorrectly states "ROM sets can be of two types: merged or splitted. In a merged set, parent ROMs are merged into chidren ROMs, resulting in a waste of disk space and the ability to launch a child ROM without the presence of its parent."

I think this is a combination of a factual mistake (there are three types of ROM sets) and maybe a language error in the "waste of disk space" sentence.

I no longer have access to the wiki, so I'd like to suggest this explanation, borrowed from other sources:

ROM sets can be of three types: Non-Merged, Split, and Merged. The question of merged, split, and non-merged ROM sets has to do with "parent" and "clone" games. The parent is the primary or original version of a game. With all varieties (merged, split, non-merged) of MAME ROM sets, the parent ROM has all the files needed to play them within the .zip file. Clones are a variant of the Parent. 

Merged" and "Split" sets save drive space by consolidating the parts of the clone game that are identical to its parent into a more sophisticated zip container. Non-Merged sets arrange all games, parents and clones, in their own stand-alone .zip files with all of the files needed to run the game. This takes somewhat more storage space but removes clone dependencies on parents, meaning you can take any of ROM files (including clones), put them in the appropriate emulator ROM directory on their own, and they should work.

Kivutar, 2016-08-15:
But larrykoubiak told me the opposite. He said that standalone ROMs are called merged because to achieve a standalone ROM, you have to merge some file from the parent in the child.

I don't get the difference between Merged and Split in your explanation.

Kivutar, 2016-08-15:
I'm reading this http://mamedev.emulab.it/clrmamepro/docs/htm/merger.htm

markwkidd, 2016-08-15:
I guess my definition is not as clear as it should be yet either.  :)

I would say that in general Non-Merged sets should be preferred because each ROM file works on its own without having the parent ROM. This can save a lot of time and effort for users who are often not interested in having every single arcade game, but instead would like to have just a list of games that they enjoy.

@dankcushions has done a lot of work helping users try to get their MAME ROMs running in RetroArch and in RetroPie. Maybe dankcushions can help edit the text a bit more to make it clar. (Unfortunately I don't seem to be able to tag dankcushions in this ticket since they aren't a part of this repository. I'll try to get their attention another way)

Kivutar, 2016-08-15:
I understand the definitions now. And I updated the wiki page. Can you check that I didn't say anything wrong please?

I'm also for Non-Merged since I just play a few ROMs, but @RobLoach and @larrykoubiak who are using full sets are for Splited set.

dankcushions, 2016-08-15:
yeah agreed that there are use-cases for all the different set types. I used non-merged sets but I don't use full sets or current mame. a non-merged full set of current mame would be HUGE! :)

it is difficult to explain the set types in a concise way but happy to review anything if anyone could supply the link?

Kivutar, 2016-08-15:
Thanks! Here is the link: https://github.com/libretro/Lakka/wiki/Arcade

dankcushions, 2016-08-15:
reads good to me! you could go into more detail about what a parent/clone rom is but i think you can end up making a long guide about MAME roms and those never end :)

markwkidd, 2016-08-15:
Thank you for making those changes, it seems much clearer now.

I think there is a myth about how much space that Merged and Split sets save. Those are somewhat useful tactics for saving drive space, but when compared to the size of a PSX set for example, the 'gains' of splitting and merging MAME sets don't seem as impressive.

For example, complete (ROM+CHD) MAME 0.139. Non-Merged set: 298GB, Split set: 222GB (~25% reduction in size)

That's a thought for another time though.

closed 2016-08-21 by markwkidd