libretro/Lakka #465: Flower Screen

alexandrebfaust, 2016-08-16:
**My Lakka not working...**

Only is flower screen, I can select Installer or Live 

I tried two modes

[image]

gouchi, 2016-08-16:
Please can you provide some [logs](https://github.com/libretro/Lakka/wiki/Troubleshooting-Lakka).

The more information we will have, the easiest it will be to help you.

Thank you.

alexandrebfaust, 2016-08-16:
Is it?

[image]
[image]

gouchi, 2016-08-16:
Thank you, it doesn't seem to have fatal error.

Can you provide the output of 
`lspci -nn | grep VGA`

You are using which video output on which hardware ?

Kivutar, 2016-08-16:
Are you using two monitors?

alexandrebfaust, 2016-08-17:
I'm using a notebook without screen
cce win wm545b
[image]
[image]

gouchi, 2016-08-17:
Try to change [video_monitor_index](https://github.com/libretro/RetroArch/blob/master/retroarch.cfg#L149) in the [configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface) as it seems you are not using the main display.

Nano shortcuts are CTRL+O (Save), CTRL+W (Search), CTRL+X (Quit)

alexandrebfaust, 2016-08-18:
OOOH YEAH BRO! It's working! :+1: If I restart my pc in live mode, his continue working with this edited file?

alexandrebfaust, 2016-08-18:
Now the question is: How I connect in a wifi network? :smile:

cudencuden, 2016-08-18:
http://www.lakka.tv/doc/Network-settings/

alexandrebfaust, 2016-08-18:
Thanks! :+1: :+1: :+1: :+1:

closed 2016-08-18 by alexandrebfaust