libretro/Lakka #466: RPi3 can't see wifi

duramato, 2016-08-16:
After following every step mentioned on the wiki on how to connect to a **protected** wifi , my AP doesn't appear listed

```
Lakka:~ # connmanctl enable wifi
Error wifi: Already enabled
Lakka:~ # connmanctl
connmanctl> agent on
Agent registered
connmanctl> scan wifi
Scan completed for wifi
connmanctl> services
*AR Wired                ethernet_REDACTED_cable
    Chromecast Da Sala   wifi_REDACTED_managed_none
    MEO-CASA             wifi_REDACTED_managed_psk
    MEO-WiFi             wifi_REDACTED_managed_none
    MEO-7C0B47           wifi_REDACTED_managed_psk
    ZON-NOS              wifi_REDACTED_managed_psk
    Vodafone-EBA97B      wifi_REDACTED2_managed_psk
    FON_ZON_FREE_INTERNET wifi_REDACTED_managed_none
connmanctl>
```

Mine is called `Matos Home` and as you can see it is not there. At first i tought it was becouse of an apostrophe i had `Matos' Home` so i changed it to what it is now, same thing, i then questioned myself if it is becosue of the space in the name but `Chromecast Da Sala` has spaces and it is shown
@Kivutar follow up of the irc discussion

Kivutar, 2016-08-18:
Does your AP have something special that makes it not appear there? This is going to be difficult to debug.

duramato, 2016-08-21:
I don't think so, feel free to ask me about any details of the AP as i don't know what to say of it.

Kivutar, 2016-08-25:
Does the same bug affects OpenELEC?

spec-chum, 2016-09-02:
Just to add to this, I got wi-fi working on my Pi 3, but also remember the Pi 3 doesn't have 5Ghz wi-fi so if Matos Home is 5Ghz it's not going to show.

Kivutar, 2016-10-02:
@duramato could it be the cause of the bug?

duramato, 2016-10-02:
I haven't tried with OpenELEC yet, but no it isn't 5ghz, it's 2.4ghz

Kivutar, 2016-10-04:
Sorry, I have no idea how to debug this. Testing it on openelec would help us to know if there is a quick fix available.

Kivutar, 2016-12-16:
I have no way to debug this without you testing OpenELEC

Kivutar, 2017-02-06:
Try the libreelec based images please http://le.builds.lakka.tv/