libretro/Lakka #467: allow for fuzzy logic while scanning roms / creating playlists

trapexit, 2016-08-16:
If enabled I'd imagine the scanning would go as follows.

1. Check the internal DB
2. Normalize/simplify the ROM filename (remove symbols, common metadata) and check the internal database names for the one which has the highest statistical similarity. (Remove symbols from filename and then use something like python's difflib's SequenceMatcher).
3. Cleanup the filename and use that as the name.

Similar heuristics are used by some movie, tv, & music naming tools.

cudencuden, 2016-08-16:
i think currently supported scanning > playlist creation is based on serial number or crc or md5....is this for any system that currently do not support playlist creationg via scanning?

trapexit, 2016-08-16:
Yes and in situations where the ROM isn't matched for whatever reason.

The casual end user doesn't care about the "purity" of the ROM or some minor differences in the release. It'd be nice to indicate that a ROM is not the known good one but I still want it to show up in the playlist without me having to manually create the entries or trying to track down the "good" version.

Filenames are decent sources of metadata... why not use them?

cudencuden, 2016-08-17:
yeah, other brought this up in the past...other questions with no good implementation is how will lakka know which system to put the fuzzy match under....for now, i would put all of your non-recognized in one folder and look at the scripting section in http://www.lakka.tv/doc/Playlists/

trapexit, 2016-08-18:
It could infer based on certain info. The extension, directory name they are in, the closest match in existing DBs, and or just ask.

Kivutar, 2016-08-18:
Extension based scanning is too fuzzy. People were complaining and complaining again one year ago when we had that fuzzy scanning based on extensions. There are use cases for which it doesn't work.

Same for name based scanning. Some people have roms called FF6.SMC, I don't see how we can match this with "Final Fantasy VI (Japan)".

In our documentation, you will find a 10 lines script that can generate a playlist based on a folder and a wildcard pattern. However, that doesn't link to the database entries.

The current system works perfectly if you are using the No-Intro sets, the official FBA set, the official MAME set, and soon REDUMP for the CD images.

And maintaining two ways of scanning make our code messier.

trapexit, 2016-08-18:
Extensions are just 1 of several pieces of metadata.Using it exclusively is wrong but there are ways to use it to create a more complicated heuristic.

"Same for name based scanning. Some people have roms called FF6.SMC, I don't see how we can match this with "Final Fantasy VI (Japan)"."

By codifying how a human would. "FF" is probably an abbreviation and it's not uncommon to use roman numerals swapped with arabic. If you were to do a simple wildcarding on names you can have something like "F* F* 6*" and "F* F* VI*". 

The final algo would be something like:
1. Check hash
2. If well known extension (or directory name matches a common system name)
2a. check DB of that system for filename.
2b. check DB of that system for wildcarded abbreviations
2b. check DB of that system for certain well known shortened or substituted words/numbers
3. Expand 2 to all DBs
4. Possibly use Levenshtein distance or similar to compare the name against the known

closed 2016-08-18 by trapexit

cudencuden, 2016-08-18:
but the fuzzy database of possible name matches will become huge....it is not worth it, i dont think

trapexit, 2016-08-18:
The DB would only hold the primary name and misc well known metadata as today. The fuzziness is in the interpretation for performing searches. Literal searches for things like hashes and filesize (perhaps a bit fuzziness on size). Some smarts around handling filenames, extensions and parent directory names as hints to the system.