libretro/Lakka #468: allow case insensitivity when looking for files referenced from a cue file

trapexit, 2016-08-16:
Not an uncommon problem to have CUEs refer to wrongly cased files.

sergiobenrocha2, 2016-09-07:
Get the .cue from redump

trapexit, 2016-09-07:
I now realize this is not the correct repo for this request but:

Why complicate the user's life? The change would be rather trivial.

Alcaro, 2016-09-07:
> The change would be rather trivial.

You sure? Enlighten me. The only method I'm aware of is 'enumerate the directory and check every filename', which is far from trivial.

trapexit, 2016-09-07:
In what way is `opendir, loop on readdir + case insensitive strcmp, closedir, return actual proper filename` complicated? It's maybe 10 lines of code.

sergiobenrocha2, 2016-09-07:
Then do it yourself... but I think it's better to keep with standard .cue instead the broken ones

trapexit, 2016-09-07:
Anyone know where the code lives? I'm happy to do it. 

Those cue files are not broken on Windows, OSX, or FAT filesystems on Linux and simply confuses users given there is no feedback to when something fails.

Kivutar, 2016-09-07:
Take a look at task_database.c

trapexit, 2016-09-09:
Thanks. I was expecting the the cue parsing logic to be normalized but it exists in each of the cores. So while changing the database code would allow it to search and index in a case insensitive way each core would need to change to get them to load the files. 

Alternatively perhaps tooling could be created and included to Lakka or retroarch to help discover and fix such issues. Ideally the UI and cores could provide a better error reporting API.

closed 2016-09-09 by trapexit