libretro/Lakka #471: Dell Optiplex GX260 Flickers Flower Screen

Kaizodude, 2016-08-21:
So First off, I'm new to within so you'll have to excuse me for not writing in the right manner. 
My issue is the following
My screen keeps flickering the image of the flower. I am using a Dell Optiplex GX260 with 1GB Ram , 20GB HDD and a Intel pentium 4 at 2GHz. 
i have tried looking on the BIOS +really old one+ and no settings changed it. I will post a quick video of the issue

Kaizodude, 2016-08-21:
https://youtu.be/flnW7jfYkWo

gouchi, 2016-08-21:
Do you reproduce the issue with [latest nightly](http://mirror.lakka.tv/nightly/Generic.i386/Lakka-Generic.i386-devel-20160811142808-r21565-gc79651a.img.gz) ?

Kaizodude, 2016-08-21:
Well.. I know it's a nightly build. It's i386 and it was for PC and it was from August 11th 2016

Kaizodude, 2016-08-21:
So yes. The latest nightly build for 32bit PC
http://sources.lakka.tv/nightly/Generic.i386/

Kaizodude, 2016-08-21:
Also. As of right now. When I click installer it lags alot. Dosent get past flower

gouchi, 2016-08-21:
Can you try to boot by adding to [boot arguments](http://wiki.openelec.tv/index.php/Boot_Arguments) (Press Tab to edit cmdline) **textmode** then try to get some [log](http://www.lakka.tv/doc/Troubleshooting-Lakka) ?

Thank you.

Kaizodude, 2016-08-21:
OK one moment. Wherever I try and boot it does this https://youtu.be/TdbDdPqNoWY
It just sticks at the flower screen, even when I type in the command

gouchi, 2016-08-21:
Try to add boot arguments for Live Mode.

Can you try to press F12 to access Boot Menu and boot on your USB key ?

Kaizodude, 2016-08-21:
That's what I've been doing. Sorry for the confusion

gouchi, 2016-08-21:
Another thing you can try is to boot first with [Plop Boot Manager on CD](https://www.plop.at/en/bootmanager/plpbt.bin.html#runcd) to boot to the usb key.

Maybe it will help.

Kaizodude, 2016-08-21:
Ok ok hold on. My Dell is lagging alot right now. It's loading 1 line of pixels per second... I don't know what's going on

Kaizodude, 2016-08-21:
Opps

Kaizodude, 2016-08-21:
Ah one moment

Kaizodude, 2016-08-21:
How do I get the logs?
All I see is the blinking flower

Kaizodude, 2016-08-21:
Location

Kaizodude, 2016-08-21:
Never mind I found it. I will post a pic

Kaizodude, 2016-08-21:
Uhg. Sorry for being a pain
 I revoke that nevermind. Where do I find it again?

Kaizodude, 2016-08-21:
And what are those I things your putting?

Kaizodude, 2016-08-21:
Ok so I know where to put the commands. It only lets me type the first and third one in. The 2nd crashes it

Kaizodude, 2016-08-21:
But where's the output file?

gouchi, 2016-08-21:
Try to come on [IRC](http://webchat.freenode.net/?channels=%23lakkatv) so that other people can also help you with the reference of this [issue](https://github.com/libretro/Lakka/issues/471).

Kaizodude, 2016-08-21:
No one appears to be online :/

gouchi, 2016-08-21:
Graphic card (8086:2562) is too old to handle RetroArch with kernel 4.6

closed 2016-08-21 by gouchi