libretro/Lakka #473: Lakka damaging hds data

naaando, 2016-08-23:
I had to reinstall my system twice times due to write mode of Lakka on hard disks conected on the same device. Write mode for all disks ins`t safe and therefore shouldnt be actived by default. I had same problems on Lakka partitions few times but now it`s damaging data outside of its driver too.
[image]

naaando, 2016-08-23:
More Info:
build : 	Lakka-Generic.x86_64-devel-20160811142524-r21565-gc79651a.img.gz
CPU : AMD FX(tm)-6300 Six-Core Processor × 6 
Ram : 4gb

How happened: 
I downloaded the img, wrote in usb using dd, restarted the pc, Lakka resized partition and when rebooted it said has no bootable device.

Kivutar, 2016-08-24:
Try to add udevil=0 to the cmdline

naaando, 2016-08-25:
Seems to work, no problem til now, thank you, btw, how can I help at translation? My english isn't the best but I can translate few strings(or all of then anyway).

Kivutar, 2016-08-25:
Ok, I think I should do something to enable/disable udevil from the GUI.

Translation RetroArch is not very easy for now, if you don't know how to use git and code. The translations are directly in the C code.

naaando, 2016-08-25:
I'll take a look this weekend!

Kivutar, 2016-12-20:
I was able to reproduce this bug.
I booted lakka a first time in live mode.
udevil mounted my drives and I scanned some roms.
Then I rebooted.
No way to access the drives anymore, and when I do a `ls` I get a `Structure needs cleaning` error.