libretro/Lakka #474: Lakka 64-Bit for PC Problem - FBA Emulator Gone?

Unadan, 2016-08-23:
I just upgraded to the latest nightly for my 64-bit PC (23-Aug-2016 05:53) and the FBA emulator seems to be gone from Lakka.  When I attempt to open an FBA rom I don't even get an option to open the game with FBA.  Please help!!

gouchi, 2016-08-25:
The latest nightly is from [11 August](http://mirror.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160811142524-r21565-gc79651a.img.gz). 
Can you show the content of FB Alpha - Arcade Games.lpl ?

It should be fixed with [this commit](https://github.com/libretro/Lakka/commit/9fc2310fd58c35a2cb056ae722623815c244a6b5).

Kivutar, 2016-08-25:
It's my fails, I pushed a build on 23, then noticed that the core had been renamed. So I removed this build from the server. You can simply donwgrade to the build of august 11.

Kivutar, 2016-08-25:
This issue is fixed in our repo, so I close it.

closed 2016-08-25 by Kivutar