libretro/Lakka #477: ODROID XU4 - cannot set resolution

hooby3dfx, 2016-08-30:
I have a 720p native resolution TV which also accepts 1080i input.

The Lakka image always seems to boot to 1080i regardless of whether I follow the instructions to update the boot.ini file. This results in poor performance at 30 Hz. 
I would like to force it to 720p.
My TV is a little funky as described here http://libretro.com/forums/showthread.php?t=6803&p=46164 however the official ODROID Android image correctly boots to 720p.

Possible duplicate of https://github.com/libretro/Lakka/issues/261

Outside of the boot.ini file, is there any other way to set the resolution? The retroarch fullscreen settings in the cfg dont seem to work either.

hooby3dfx, 2016-08-30:
Oh WTF!
The last line of the included boot.ini does not pass the video_output param!!
Hence following the instructions would have no effect XD
```
#------------------------------------------------------------------------------------------------------
#
# Kernel boot
#
#------------------------------------------------------------------------------------------------------
setenv bootargs "${console} ${bootrootfs} ${openelec} smsc95xx.macaddr=${macaddr}"
```
should be
```
#------------------------------------------------------------------------------------------------------
#
# Kernel boot
#
#------------------------------------------------------------------------------------------------------
setenv bootargs "${console} ${bootrootfs} ${video_output} ${openelec} smsc95xx.macaddr=${macaddr}"

```

SO, now that I can actually set the resolution I am getting this type of output:
[image]

cudencuden, 2016-08-30:
for retroarch.cfg changes, did you stop retroarch service before making changes? systemctl stop retroarch

Kivutar, 2016-08-30:
Ah you're right! Well spotted. I need to change this. I will do just after I'm finished with the LibreELEC port.

Sadly I have no idea why you are getting this kind of signal.

Can you try the fbset command line in this script? https://github.com/zalaare/OpenELEC-Odroid/blob/odroid-6.0/projects/Odroid/devices/C1/initramfs/platform_init

hooby3dfx, 2016-08-30:
> 0/projects/Odroid/devices/C1/initramfs/platform_init

I found something else interesting. I dont think the issue is with my TV.
After getting the 720p mode set, I plugged the odroid into my other
display, which also got the distorted 720p output.

I think there is an issue with the Lakka image causing 720p output to have
the problem all the time on the xu4 at least (don't have an xu3).

Can you reproduce it now by forcing 720p with the fixed boot.ini?

I will also try and fiddle with fbset; which command in particular?

Kivutar, 2016-08-31:
Could be related http://forum.odroid.com/viewtopic.php?f=94&t=15840

hooby3dfx, 2016-08-31:
But that doesn't explain why it would work on Android?

hooby3dfx, 2016-09-01:
Can you point me in the direction of the code/script that handles initializing the settings for the graphics mode in Lakka (at least for odroid xu3/4)?

hooby3dfx, 2016-09-01:
Ah ha! Here is another thread of someone else with what looks like the same issue from the photo:
http://forum.odroid.com/viewtopic.php?f=99&t=22716

As a temporary work around, it looks like 1024x768 works.

Kivutar, 2016-09-13:
We do have the commit he's talking about in that thread.

So it's a triple buffering issue?

hooby3dfx, 2016-10-21:
What do you think the next step is here?

If anyone else with an ODROID XU4 sees this, can you force 720p and see if you get the same output issue on your TV/monitor?

Kivutar, 2016-10-21:
I think the next step is to experiment with fbset.

You have some examples here in the C1 project https://github.com/libretro/Lakka/blob/lakka/projects/OdroidC1/initramfs/platform_init

hooby3dfx, 2016-10-23:
Is there a way confirm the kernel source (like the commit hash) that was used?
The latest stable Lakka build outputs this for the version:
```
Lakka:~ # uname -a
Linux Lakka 3.10.96 #1 SMP PREEMPT Wed Sep 14 06:27:12 CEST 2016 armv7l GNU/Linux
Lakka:~ # cat /proc/version
Linux version 3.10.96 (kivutar@gcc20) (gcc version 4.9.2 (GCC) ) #1 SMP PREEMPT Wed Sep 14 06:27:12 CEST 2016
```
Shouldn't that be higher?

Kivutar, 2016-10-23:
The commit id https://github.com/libretro/Lakka/blob/lakka/packages/linux/package.mk#L34

hooby3dfx, 2016-10-26:
Thanks - a couple updates:
1. I found that the issue does not seem to occur on OpenELEC-6.0.2.0 (Linux OpenELEC 3.10.82+30cd824 from http://forum.odroid.com/viewtopic.php?f=79&t=5915)
2. The issue appears during boot of the official hardkernel linux release (3.10.103; http://odroid.com/dokuwiki/doku.php?id=en:xu3_ubuntu_release_note_20161011)
3. Issue occurs in Lakka build as far back as Lakka-OdroidXU3.arm-devel-20160109151244-r21276-g4dadcab

Enverex, 2016-12-28:
I'm getting the same when trying to use a protector on my Odroid-XU4 with Lakka (the projector's native res is 1280x800). Seems to happen at 720p too (forced via boot.ini).

Is this basically unusable except at 1080p?

hooby3dfx, 2016-12-28:
That is what I believe.

hooby3dfx, 2016-12-29:
@Enverex the issue seems to just be with 720p. a temp workaround is actually to force 1024x768 via the boot file.

Update on this issue - after a little more testing, I confirmed that the problem exists upstream and even on the current official odroid Ubuntu release.
It only seems to occur with 720p when using fbdev. I will be reporting the bug shortly and once fixed Lakka can pull it in.

hooby3dfx, 2017-01-01:
Fixed upstream!
https://github.com/hardkernel/linux/commit/bc94be360ae2652484afbd2c4a61b55b0a6d53ad

Enverex, 2017-01-02:
I don't think that would fix it for me as I'm using 1280x800 and they seem to be hard-coding specific resolutions in... which seems an odd way of doing it.

hooby3dfx, 2017-01-02:
Can you chime in on the odroid forum?

hooby3dfx, 2017-01-02:
well at least this fixes 720p, which i think a lot of users have basically been blocked by.
@Kivutar can this go on your radar to pull in?

Kivutar, 2017-01-15:
Yes, i'm updating it.