libretro/Lakka #481: Stuck at flower flashing on and off

malta108, 2016-09-09:
Looking at the log at the bottom it says:

`RetroArch [ERROR] :: Failed to compile fragment shader #0
RetroArch [ERROR] :: Failed to link program #0.
RetroArch [ERROR] :: GLSL stock programs failed to compile.
RetroArch [ERROR] :: [GL]: Shader initialization failed.`

gouchi, 2016-09-09:
Which graphic card do you have ?

Is it possible to get output of 
`lspci -nn | grep VGA`

Thank you

malta108, 2016-09-10:
I get `VGA compatible controller [0300]: Intel Corporation 82865G Integrated Graphics Controller [8086:2572] (rev 02)`

gouchi, 2016-09-11:
I forgot some options sorry can you try with
lspci -nnk | grep -A 3 VGA

But I am guessing you are using a graphic card which is too old for the kernel we are using now.

Also, you can try to change the menu driver to rgui by editing the [configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface).

Search the entry `menu_driver = "xmb"` and change it to `menu_driver = "rgui"`

malta108, 2016-09-12:
I ran `lspci -nnk | grep -A 3 VGA`
the result was `VGA compatible controller [0300]: Intel Corporation 82865G Integrated Graphics Controller [8086:2572] (rev 02)`
`Subsystem: ASRock Incorporation 82865G Integrated Graphics Controller [1849:2572]`
`Kernel driver in use: i915`
`System preipheral [0880]: Intel Corporation 82865G/PE/P Processor to I/O Memory Interface [8086:2576] (rev 02)`

I also tried disabling ribbon the option wasn't there but I added it anyway it didn't work, same error.

I changed the menu driver and it still didn't work.

gouchi, 2016-09-12:
Keep rgui menu driver and try changing your video_driver from `gl`  to `sdl2` or `xvideo` in the configuration file.

malta108, 2016-09-12:
the video_driver wasn't there I added it but it didn't change anything.

malta108, 2016-09-12:
I got the log while opening sdl2
`RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 
Built: Aug 11 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 285aa2b
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [ERROR] :: Couldn't find any video driver named "sdl2"
RetroArch [INFO] :: Available video drivers are:
RetroArch [INFO] :: 	gl
RetroArch [INFO] :: 	null
RetroArch [WARN] :: Going to default to first video driver...
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29975.00 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [ERROR] :: Couldn't find any video driver named "sdl2"
RetroArch [INFO] :: Available video drivers are:
RetroArch [INFO] :: 	gl
RetroArch [INFO] :: 	null
RetroArch [WARN] :: Going to default to first video driver...
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 1 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 11 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Mode 0: (1280x1024) 1280 x 1024, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (1280x1024) 1280 x 1024, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (1152x864) 1152 x 864, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (1024x768) 1024 x 768, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 5: (832x624) 832 x 624, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 6: (800x600) 800 x 600, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 7: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 8: (640x480) 640 x 480, 75 Hz
RetroArch [INFO] :: [DRM]: Mode 9: (640x480) 640 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 10: (720x400) 720 x 400, 70 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1280x1024.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: [EGL]: Current context: 0x7f492800c260.
RetroArch [INFO] :: [KMS]: New FB: 1280x1024 (stride: 8192).
RetroArch [INFO] :: [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) 865G .
RetroArch [INFO] :: [GL]: Version: 1.3 Mesa 11.2.0-rc1.
RetroArch [INFO] :: Querying GL extension: ARB_ES2_compatibility => doesn't exist
RetroArch [INFO] :: Querying GL extension: ARB_texture_non_power_of_two => doesn't exist
RetroArch [INFO] :: Querying GL extension: ARB_fragment_program => doesn't exist
RetroArch [INFO] :: Querying GL extension: ARB_sync => exists
RetroArch [INFO] :: Querying GL extension: ARB_texture_float => doesn't exist
RetroArch [INFO] :: Querying GL extension: EXT_texture_sRGB => doesn't exist
RetroArch [INFO] :: GL: Using resolution 1280x1024
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [ERROR] :: Failed to compile fragment shader #0
RetroArch [ERROR] :: Failed to link program #0.
RetroArch [ERROR] :: GLSL stock programs failed to compile.
RetroArch [ERROR] :: [GL]: Failed to initialize shader, falling back to stock.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [ERROR] :: Failed to compile fragment shader #0
RetroArch [ERROR] :: Failed to link program #0.
RetroArch [ERROR] :: GLSL stock programs failed to compile.
RetroArch [ERROR] :: [GL]: Shader initialization failed.
Segmentation fault
`

gouchi, 2016-09-12:
I think your graphic card is too old to be supported :(

malta108, 2016-09-12:
ok thanks anyway for the information.

closed 2016-09-12 by gouchi