libretro/Lakka #483: First boot can't into system

gaozp, 2016-09-12:
I use raspberry B+ , and I follow the step of website,  when I got to the step of first boot , my screen is gray ,and there is no flow splash screen , I was waiting for about 5 minutes . 

Do I need to catch some log for you guys?

gouchi, 2016-09-12:
Yes, thank you that will be great. Please check [this documentation](http://www.lakka.tv/doc/Troubleshooting-Lakka).

Did you check if the checksum was ok  [md5](http://mirror.lakka.tv/nightly/RPi.arm/Lakka-RPi.arm-devel-20160811144835-r21565-gc79651a.img.gz.md5) or [sha256](http://mirror.lakka.tv/nightly/RPi.arm/Lakka-RPi.arm-devel-20160811144835-r21565-gc79651a.img.gz.sha256) ?

gaozp, 2016-09-12:
The image downloaded from website's full name is **Lakka-RPi2.arm-devel-20160811145614-r21565-gc79651a.img** and MD5 is:
MD5 (Lakka-RPi2.arm-devel-20160811145614-r21565-gc79651a.img) = 9e5969135a5f6c9aa4e5a6ca9a9ddb1c

I'll follow the documentation and provide the logs.

gouchi, 2016-09-12:
If you have b+  one you should use [RPi image](http://mirror.lakka.tv/nightly/RPi.arm/Lakka-RPi.arm-devel-20160811144835-r21565-gc79651a.img.gz) but if you have 2B one you should use [RPi2 image](http://sources.lakka.tv/nightly/RPi2.arm/Lakka-RPi2.arm-devel-20160811145614-r21565-gc79651a.img.gz).

gaozp, 2016-09-14:
@gouchi I have solved this problem. 
There are two wired things concerned with this 
1.My pi named B+ but I chose RPi image.
2.The website said the size of the RPi image is 128MB, and when I downloaded , the size is 574.6MB.

I'll close this issue , and if you need some screenshots or photos, I'll post some .

Thank you all the same :)

closed 2016-09-14 by gaozp