libretro/Lakka #485: XMB UI text disappears in Lakka when scrolling through huge collections

OGWillikers, 2016-09-18:
I've got a huge collection on a network drive, running on an RPi 3 and screenshot thumbnails enabled.  If I go scrolling through a list of games for about 5-10 minutes, the thumbnail disappears, and a few more scrolls later all text on the screen starts flickering and disappears.  Demanding games seem to perform poorly when this occurs, if I can blindly launch one.

Disabling thumbnails either stops or greatly increases the time to failure (has only happened once on an older build that I've seen) and using boxart thumbnails greatly decreases time.

gouchi, 2016-09-20:
Thank you for your report. Is it possible to have a video or a picture of the issue ?

OGWillikers, 2016-09-20:
I'll get on it.  Gotta rescan my collection since I moved some things.

Kivutar, 2016-09-20:
Looks like the bug found by @lollo78

OGWillikers, 2016-09-20:
Yeah, that sounds the same.  Apologies for the duplicate.  I can still create a video if it'd be useful.  Looks like the video posted on the other issue is down.

Kivutar, 2016-09-21:
It's difficult to find the cause of this bug, since it doesn't happen on PC.

Kivutar, 2016-11-15:
Can you try to increase the memory attributed to the gpu in config.txt?

Try to set `gpu_mem_1024=512`

lollo78, 2016-11-15:
I'll die before to see this bug fixed. 😂