libretro/Lakka #486: Stuck at flower screen on Amd CPU 2015 and Intel CPU 2011 model

geekyjithin, 2016-09-19:
Tried live mode with latest stable build on 2 notebook amd and Intel CPU. Stuck at flower screen 
1.HP 15-af001AU 39.62 cm 2015 model
AMD APU Quad Core A6- 6310
4gb ram
AMD Radeon R4 Graphics processor 

2. Dell Inspiron 15r N5110
Processor2nd generation Intel® Core™ i3-2310M Processor (2.1GHz, Dual Core, 4T, 3MB L3)
Memory3GB  DIMM DDR3 1333Mhz
Video CardNVIDIA GeForce GT 525M - 1GB
Hard Drive320GB 5400RPM Hard Drive

gouchi, 2016-09-19:
Is it possible to get [Lakka system](http://www.lakka.tv/doc/Troubleshooting-Lakka/#lakka-system-logs) and [graphic card](http://www.lakka.tv/doc/Troubleshooting-Lakka/#graphic-card-logs) logs ?

You can try with the [latest nightly](http://sources.lakka.tv/nightly/Generic.x86_64/Lakka-Generic.x86_64-devel-20160917123350-r21587-g0fb570f.img.gz) available.

Thank you.

geekyjithin, 2016-09-19:
I also tried with latest nightly build...
But the results are same...
I am not a tech genius.. excuse me for my ignorance....
How to get system and graphics card log , when it is stuck at the flower screen... only thing I can do is restarting the system.....
Thank you...

GeoDragon30, 2016-09-19:
Do you have a monitor hooked up to the notebook? That was a problem I had. The monitors was stuck on the flower screen but the notebook had the menu.

gouchi, 2016-09-19:
@geekyjithin to get the log you need to access [lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface).

geekyjithin, 2016-09-19:
My amd notebook is not with me right now
I am using my old laptop with a CRT monitor since it had a broken display..
I also connected with my TV with HDMI cable result is same

RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX 
Built: Sep 16 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 0fb570f
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg.
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565.
RetroArch [INFO] :: Version of libretro API: 1
RetroArch [INFO] :: Compiled against API: 1
RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz.
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp"
RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl"
RetroArch [INFO] :: Video @ 960x720
RetroArch [INFO] :: Starting threaded video driver ...
RetroArch [INFO] :: [DRM]: Found 2 connectors.
RetroArch [INFO] :: [DRM]: Connector 0 connected: yes
RetroArch [INFO] :: [DRM]: Connector 0 has 5 modes.
RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1.
RetroArch [INFO] :: [DRM]: Connector 1 connected: no
RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes.
RetroArch [INFO] :: [DRM]: Mode 0: (1024x768) 1024 x 768, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 1: (800x600) 800 x 600, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 2: (800x600) 800 x 600, 56 Hz
RetroArch [INFO] :: [DRM]: Mode 3: (848x480) 848 x 480, 60 Hz
RetroArch [INFO] :: [DRM]: Mode 4: (640x480) 640 x 480, 60 Hz
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
libGL: Can't open configuration file /storage/.drirc: No such file or directory.
RetroArch [INFO] :: Found GL context: kms
RetroArch [INFO] :: Detecting screen resolution 1024x768.
RetroArch [INFO] :: [EGL]: EGL version: 1.4
RetroArch [INFO] :: [EGL]: Current context: 0x7f773802a5f0.
RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096).
RetroArch [INFO] :: [GL]: Vendor: nouveau, Renderer: Gallium 0.4 on NVC1.
RetroArch [INFO] :: [GL]: Version: 3.0 Mesa 11.2.0-rc1.
RetroArch [INFO] :: GL: Using resolution 1024x768
RetroArch [INFO] :: [GL]: Default shader backend found: glsl.
RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend.
RetroArch [INFO] :: Checking GLSL shader support ...
RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: Found GLSL vertex shader.
RetroArch [INFO] :: Found GLSL fragment shader.
RetroArch [INFO] :: Linking GLSL program.
RetroArch [INFO] :: [GL]: Using 4 textures.
RetroArch [INFO] :: [GL]: Loaded 1 program(s).
RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
RetroArch [INFO] :: [udev] Adding device /dev/input/event7 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event6 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev] Adding device /dev/input/event9 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_MOUSE.
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 1024 frames
RetroArch [INFO] :: ALSA: Buffer size: 2048 frames
RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: Using font rendering backend: freetype.
function is no-op
RetroArch [INFO] :: Using font rendering backend: freetype.
function is no-op
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096).
RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096).
function is no-op

gouchi, 2016-09-19:
Thank you for the log.

Try to change [video_monitor_index](https://github.com/libretro/RetroArch/blob/master/retroarch.cfg#L157) in the [configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface) with
`video_monitor_index = 1`

Nano shortcuts are CTRL+O (Save), CTRL+W (Search), CTRL+X (Quit)

geekyjithin, 2016-09-21:
Tried but failed....
Tested latest retroarch 1.3.6 on the same system on Windows 7 worked smoothly....so problem is not with graphics card i think..
I am attaching log and retroarch config file...
RetroArch [INFO] :: === Build ======================================= Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX Built: Sep 16 2016 RetroArch [INFO] :: Version: 1.3.6 RetroArch [INFO] :: Git: 0fb570f RetroArch [INFO] :: ================================================= RetroArch [INFO] :: Looking for config in: "/storage/.config/retroarch/retroarch.cfg". RetroArch [INFO] :: Config: loading config from: /storage/.config/retroarch/retroarch.cfg. RetroArch [INFO] :: Resetting undo buffers. RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: RGB565. RetroArch [INFO] :: Version of libretro API: 1 RetroArch [INFO] :: Compiled against API: 1 RetroArch [INFO] :: Set audio input rate to: 29970.03 Hz. RetroArch [INFO] :: Found shader "/usr/share/common-shaders/bilinear.glslp" RetroArch [INFO] :: Found shader "/usr/share/common-shaders/nearest.glslp" RetroArch [INFO] :: Found shader "/usr/share/common-shaders/stock.glsl" RetroArch [INFO] :: Video @ 960x720 RetroArch [INFO] :: Starting threaded video driver ... RetroArch [INFO] :: [DRM]: Found 2 connectors. RetroArch [INFO] :: [DRM]: Connector 0 connected: yes RetroArch [INFO] :: [DRM]: Connector 0 has 5 modes. RetroArch [INFO] :: [DRM]: Connector 0 assigned to monitor index: #1. RetroArch [INFO] :: [DRM]: Connector 1 connected: no RetroArch [INFO] :: [DRM]: Connector 1 has 0 modes. RetroArch [INFO] :: [DRM]: Mode 0: (1024x768) 1024 x 768, 60 Hz RetroArch [INFO] :: [DRM]: Mode 1: (800x600) 800 x 600, 60 Hz RetroArch [INFO] :: [DRM]: Mode 2: (800x600) 800 x 600, 56 Hz RetroArch [INFO] :: [DRM]: Mode 3: (848x480) 848 x 480, 60 Hz RetroArch [INFO] :: [DRM]: Mode 4: (640x480) 640 x 480, 60 Hz RetroArch [INFO] :: Found GL context: kms RetroArch [INFO] :: Detecting screen resolution 1024x768. RetroArch [INFO] :: [EGL]: EGL version: 1.4 RetroArch [INFO] :: [EGL]: Current context: 0x7f982802a5f0. RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096). RetroArch [INFO] :: [GL]: Vendor: nouveau, Renderer: Gallium 0.4 on NVC1. RetroArch [INFO] :: [GL]: Version: 3.0 Mesa 11.2.0-rc1. RetroArch [INFO] :: GL: Using resolution 1024x768 RetroArch [INFO] :: [GL]: Default shader backend found: glsl. RetroArch [INFO] :: [Shader driver]: Using GLSL shader backend. RetroArch [INFO] :: Checking GLSL shader support ... RetroArch [WARN] :: [GL]: Stock GLSL shaders will be used. RetroArch [INFO] :: Found GLSL vertex shader. RetroArch [INFO] :: Found GLSL fragment shader. RetroArch [INFO] :: Linking GLSL program. RetroArch [INFO] :: Found GLSL vertex shader. RetroArch [INFO] :: Found GLSL fragment shader. RetroArch [INFO] :: Linking GLSL program. RetroArch [INFO] :: Found GLSL vertex shader. RetroArch [INFO] :: Found GLSL fragment shader. RetroArch [INFO] :: Linking GLSL program. RetroArch [INFO] :: Found GLSL vertex shader. RetroArch [INFO] :: Found GLSL fragment shader. RetroArch [INFO] :: Linking GLSL program. RetroArch [INFO] :: Found GLSL vertex shader. RetroArch [INFO] :: Found GLSL fragment shader. RetroArch [INFO] :: Linking GLSL program. RetroArch [INFO] :: [GL]: Using 4 textures. RetroArch [INFO] :: [GL]: Loaded 1 program(s). RetroArch [INFO] :: [GL]: Using GL_RGB565 for texture uploads. RetroArch [INFO] :: Using font rendering backend: freetype. RetroArch [INFO] :: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver. RetroArch [INFO] :: [udev] Adding device /dev/input/event7 as type ID_INPUT_KEYBOARD. RetroArch [INFO] :: [udev] Adding device /dev/input/event6 as type ID_INPUT_KEYBOARD. RetroArch [INFO] :: [udev] Adding device /dev/input/event9 as type ID_INPUT_MOUSE. RetroArch [INFO] :: [udev] Adding device /dev/input/mouse0 as type ID_INPUT_MOUSE. RetroArch [INFO] :: Found joypad driver: "udev". RetroArch [INFO] :: ALSA: Using signed 16-bit format. RetroArch [INFO] :: ALSA: Period size: 1024 frames RetroArch [INFO] :: ALSA: Buffer size: 2048 frames RetroArch [INFO] :: Found menu display driver: "menu_display_gl". RetroArch [INFO] :: Using font rendering backend: freetype. RetroArch [INFO] :: Using font rendering backend: freetype. RetroArch [INFO] :: SRAM will not be saved. RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl]. RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl]. RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl]. RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl]. RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096). RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096). RetroArch [INFO] :: Saved new config to "/storage/.config/retroarch/retroarch.cfg". RetroArch [INFO] :: Resetting undo buffers. RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_history.lpl RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_video_history.lpl RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_music_history.lpl RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_image_history.lpl RetroArch [INFO] :: Threaded video stats: Frames pushed: 2483, Frames dropped: 0. RetroArch [INFO] :: Monitor FPS estima [... cut ...] ding history file: [/storage/.config/retroarch/content_history.lpl]. RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl]. RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl]. RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl]. RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096). RetroArch [INFO] :: [KMS]: New FB: 1024x768 (stride: 4096). function is no-op

geekyjithin, 2016-09-21:
savefile_directory = "~/savefiles" savestate_directory = "~/savestates" input_remapping_directory = "~/remappings" playlist_directory = "~/playlists" libretro_directory = "/tmp/cores" libretro_info_path = "/tmp/cores" system_directory = "~/system" rgui_browser_directory = "~/roms" assets_directory = "/tmp/assets" thumbnails_directory = "~/thumbnails" rgui_show_start_screen = "false" video_threaded = "true" video_smooth = "false" video_aspect_ratio_auto = "true" video_shader_dir = "/usr/share/common-shaders" video_filter_dir = "/usr/share/video_filters" video_font_path = "/usr/share/retroarch-assets/xmb/monochrome/font.ttf" audio_driver = "alsathread" audio_filter_dir = "/usr/share/audio_filters" input_driver = "udev" input_max_users = "5" input_autodetect_enable = "true" joypad_autoconfig_dir = "/tmp/joypads" input_menu_toggle_gamepad_combo = "2" menu_driver = "xmb" menu_core_enable = "false" content_database_path = "/tmp/database/rdb" cheat_database_path = "/tmp/database/cht" screenshot_directory = "~/screenshots" video_gpu_screenshot = "false" core_assets_directory = "~/roms/downloads" playlist_names = "Atari - 2600.lpl;Atari - 5200.lpl;Atari - 7800.lpl;Atari - Jaguar.lpl;Atari - Lynx.lpl;Atari - ST.lpl;Bandai - WonderSwan Color.lpl;Bandai - WonderSwan.lpl;Cave Story.lpl;Coleco - ColecoVision.lpl;DOOM.lpl;FB Alpha - Arcade Games.lpl;GCE - Vectrex.lpl;Lutro.lpl;Magnavox - Odyssey2.lpl;MAME.lpl;Microsoft - MSX2.lpl;Microsoft - MSX.lpl;NEC - PC Engine SuperGrafx.lpl;NEC - PC Engine - TurboGrafx 16.lpl;Nintendo - Famicom Disk System.lpl;Nintendo - Game Boy Advance (e-Cards).lpl;Nintendo - Game Boy Advance.lpl;Nintendo - Game Boy Color.lpl;Nintendo - Game Boy.lpl;Nintendo - Nintendo 64.lpl;Nintendo - Nintendo DS Decrypted.lpl;Nintendo - Nintendo DS (Download Play) (BETA).lpl;Nintendo - Nintendo DSi Decrypted.lpl;Nintendo - Nintendo DSi (DLC).lpl;Nintendo - Nintendo Entertainment System.lpl;Nintendo - Satellaview.lpl;Nintendo - Super Nintendo Entertainment System.lpl;Nintendo - Virtual Boy.lpl;Quake1.lpl;ScummVM.lpl;Sega - 32X.lpl;Sega - Game Gear.lpl;Sega - Master System - Mark III.lpl;Sega - Mega Drive - Genesis.lpl;Sega - PICO.lpl;Sega - SG-1000.lpl;Sinclair - ZX Spectrum +3.lpl;SNK - Neo Geo Pocket Color.lpl;SNK - Neo Geo Pocket.lpl;Sony - PlayStation Portable.lpl;Sony - PlayStation.lpl" playlist_cores = "/tmp/cores/stella_libretro.so;DETECT;/tmp/cores/prosystem_libretro.so;/tmp/cores/virtualjaguar_libretro.so;/tmp/cores/handy_libretro.so;/tmp/cores/hatari_libretro.so;/tmp/cores/mednafen_wswan_libretro.so;/tmp/cores/mednafen_wswan_libretro.so;/tmp/cores/nxengine_libretro.so;/tmp/cores/bluemsx_libretro.so;/tmp/cores/prboom_libretro.so;/tmp/cores/fbalpha_libretro.so;/tmp/cores/vecx_libretro.so;/tmp/cores/lutro_libretro.so;/tmp/cores/o2em_libretro.so;/tmp/cores/mame_libretro.so;/tmp/cores/bluemsx_libretro.so;/tmp/cores/bluemsx_libretro.so;/tmp/cores/mednafen_supergrafx_libretro.so;/tmp/cores/mednafen_pce_fast_libretro.so;/tmp/cores/nestopia_libretro.so;/tmp/cores/mgba_libretro.so;/tmp/cores/mgba_libretro.so;/tmp/cores/gambatte_libretro.so;/tmp/cores/gambatte_libretro.so;/tmp/cores/mupen64plus_libretro.so;/tmp/cores/desmume_libretro.so;/tmp/cores/desmume_libretro.so;/tmp/cores/desmume_libretro.so;/tmp/cores/desmume_libretro.so;/tmp/cores/nestopia_libretro.so;/tmp/cores/snes9x_libretro.so;/tmp/cores/snes9x_libretro.so;/tmp/cores/mednafen_vb_libretro.so;/tmp/cores/tyrquake_libretro.so;/tmp/cores/scummvm_libretro.so;/tmp/cores/picodrive_libretro.so;/tmp/cores/genesis_plus_gx_libretro.so;/tmp/cores/genesis_plus_gx_libretro.so;/tmp/cores/genesis_plus_gx_libretro.so;/tmp/cores/picodrive_libretro.so;/tmp/cores/genesis_plus_gx_libretro.so;/tmp/cores/fuse_libretro.so;/tmp/cores/mednafen_ngp_libretro.so;/tmp/cores/mednafen_ngp_libretro.so;/tmp/cores/ppsspp_libretro.so;/tmp/cores/mednafen_psx_libretro.so" config_save_on_exit = "true" core_updater_buildbot_url = "http://buildbot.libretro.com/nightly/linux/x86_64/latest/" core_updater_buildbot_assets_url = "http://buildbot.libretro.com/assets/" content_history_path = "~/.config/retroarch/content_history.lpl" content_music_history_path = "~/.config/retroarch/content_music_history.lpl" content_video_history_path = "~/.config/retroarch/content_video_history.lpl" content_image_history_path = "~/.config/retroarch/content_image_history.lpl" cursor_directory = "~/.config/retroarch/database/cursors" dynamic_wallpapers_directory = "default" rgui_config_directory = "~/.config/retroarch/config" overlay_directory = "~/.config/retroarch/overlay" osk_overlay_directory = "~/.config/retroarch/overlay" video_driver = "gl" record_driver = "ffmpeg" camera_driver = "null" location_driver = "null" audio_resampler = "sinc" input_joypad_driver = "udev" video_aspect_ratio = "-1.000000" video_scale = "3.000000" video_refresh_rate = "59.940060" audio_rate_control_delta = "0.005000" audio_max_timing_skew = "0.050000" audio_volume = "0.000000" input_overlay_opacity = "0.700000" input_overlay_scale = "1.000000" menu_wallpaper_opacity = "0.300000" menu_footer_opacity = "1.000000" menu_header_opacity = "1.000000" video_message_pos_x = "0.050000" video_message_pos_y = "0.050000" video_font_size = "32.000000" fastforward_ratio = "0.000000" slowmotion_ratio = "3.000000" input_axis_threshold = "0.500000" input_bind_timeout = "5" input_turbo_period = "6" input_duty_cycle = "3" audio_latency = "64" audio_block_frames = "0" rewind_granularity = "1" autosave_interval = "0" libretro_log_level = "0" keyb [... cut ...] netplay_ip_address = "" recording_output_directory = "" recording_config_directory = "" core_options_path = "" video_shader = "" menu_wallpaper = "" input_overlay = "" input_osk_overlay = "" content_history_dir = "" cache_directory = "" resampler_directory = "" xmb_font = "" audio_device = "" camera_device = "" cheevos_username = "" cheevos_password = "" video_context_driver = "" input_keyboard_layout = "" bundle_assets_src_path = "" bundle_assets_dst_path = "" bundle_assets_dst_path_subdir = ""

gouchi, 2016-09-21:
When you edit the configuration file, you are sure you stopped RetroArch before editing ?

Also you can try to check [which available video outputs](http://www.lakka.tv/doc/PC/#foxconn-intel-nettop) you have and enable/disable the correct one if necessary.

By the way which video output are you trying to use ? VGA ? HDMI ?

geekyjithin, 2016-09-21:
I have edited the configuration using Linux OS before booting....
I have tried with both VGA And Hdmi.....

gouchi, 2016-09-21:
Can you try to type in Lakka ?
`for p in /sys/class/drm/*/status; do con=${p%/status}; echo -n "${con#*/card?-}: "; cat $p; done`

geekyjithin, 2016-09-22:
Thanks for your help
I finally figured out..
I have to disable both my VGA and HDMI out... in syslinux conf file.....

gouchi, 2016-09-22:
Good to know.

And you are using ? Dvi ?

geekyjithin, 2016-09-22:
No I have to disable the primary display of my laptop which was displayed as VGA -1 and HDMI port..I am using the VGA port

gouchi, 2016-09-23:
Can you paste what did you append ?
Thank you.

geekyjithin, 2016-09-23:
video=LVDS-1:d video=VGA-2:d
Thank you...

closed 2016-09-23 by gouchi