libretro/Lakka #487: Stuck at Flower Screen

Mwea, 2016-09-20:
Hi ! 
I just downloaded the latest nightly version of lakka and tried to install it on an "old" ( 2-years old ) laptop, but I get many problems.. as : 
- The only way to have access to video settings without making lakka crash is to go to the command line interface type systemctl stop retroarch and relaunch retroarch --verbose, all the other ways makes it crash 
- I can't get any sound out of my TV even if I try to change the audio device ( my laptop is connected to my tv with HDMI ) 

- If I try to launch a game, Lakka freeze on the Flower screen then crash...

Here are logs that I got when I tried to start N64probe by MooglyGuy (PD).z64.

Any help ? Thanks :+1: 

```
RetroArch [INFO] :: Using content: /storage/roms/downloads/N64probe by MooglyGuy                                                             (PD).z64.
RetroArch [INFO] :: arg #0: retroarch
RetroArch [INFO] :: arg #1: /storage/roms/downloads/N64probe by MooglyGuy (PD).z                                                            64
RetroArch [INFO] :: arg #2: -s
RetroArch [INFO] :: arg #3: /storage/savefiles
RetroArch [INFO] :: arg #4: -S
RetroArch [INFO] :: arg #5: /storage/savestates
RetroArch [INFO] :: arg #6: -c
RetroArch [INFO] :: arg #7: /storage/.config/retroarch/config/retroarch.cfg
RetroArch [INFO] :: arg #8: -L
RetroArch [INFO] :: arg #9: /tmp/cores/mupen64plus_libretro.so
RetroArch [INFO] :: arg #10: -v
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Threaded video stats: Frames pushed: 5050, Frames dropped: 0                                                            .
RetroArch [INFO] :: Monitor FPS estimation is disabled for threaded video.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/N64probe by Moo                                                            glyGuy (PD).srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/N64probe by Mo                                                            oglyGuy (PD).state".
RetroArch [INFO] :: === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 AVX AES
Built: Sep 16 2016
RetroArch [INFO] :: Version: 1.3.6
RetroArch [INFO] :: Git: 0fb570f
RetroArch [INFO] :: =================================================
RetroArch [INFO] :: Resetting undo buffers.
RetroArch [INFO] :: Loading dynamic libretro core from: "/tmp/cores/mupen64plus_                                                            libretro.so"
RetroArch [INFO] :: [overrides] no core-specific overrides found at /storage/.co                                                            nfig/retroarch/config/Mupen64Plus/Mupen64Plus.cfg.
RetroArch [INFO] :: [overrides] no game-specific overrides found at /storage/.co                                                            nfig/retroarch/config/Mupen64Plus/N64probe by MooglyGuy (PD).cfg.
RetroArch [INFO] :: Shaders: preset directory: /usr/share/common-shaders/presets
RetroArch [INFO] :: Shaders: no game-specific preset found at /usr/share/common-                                                            shaders/presets/Mupen64Plus/N64probe by MooglyGuy (PD).cgp.
RetroArch [INFO] :: Shaders: no game-specific preset found at /usr/share/common-                                                            shaders/presets/Mupen64Plus/N64probe by MooglyGuy (PD).glslp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /usr/share/common-                                                            shaders/presets/Mupen64Plus/Mupen64Plus.cgp.
RetroArch [INFO] :: Shaders: no core-specific preset found at /usr/share/common-                                                            shaders/presets/Mupen64Plus/Mupen64Plus.glslp.
RetroArch [INFO] :: Environ SET_VARIABLES.
RetroArch [INFO] :: Remaps: remap directory: /storage/remappings
RetroArch [INFO] :: Remaps: no game-specific remap found at /storage/remappings/                                                            Mupen64Plus/N64probe by MooglyGuy (PD).rmp.
RetroArch [INFO] :: Remaps: no core-specific remap found at /storage/remappings/                                                            Mupen64Plus/Mupen64Plus.rmp.
RetroArch [INFO] :: Redirecting save file to "/storage/savefiles/N64probe by Moo                                                            glyGuy (PD).srm".
RetroArch [INFO] :: Redirecting savestate to "/storage/savestates/N64probe by Mo                                                            oglyGuy (PD).state".
RetroArch [INFO] :: Environ GET_LOG_INTERFACE.
RetroArch [INFO] :: Environ GET_PERF_INTERFACE.
RetroArch [INFO] :: Environ SET_PIXEL_FORMAT: XRGB8888.
RetroArch [INFO] :: Environ GET_RUMBLE_INTERFACE.
RetroArch [INFO] :: Loading content file: /storage/roms/downloads/N64probe by Mo                                                            oglyGuy (PD).z64.
RetroArch [INFO] :: Did not find a valid content patch.
RetroArch [INFO] :: CRC32: 0x361f713f .
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-screensize:
RetroArch [INFO] ::     640x480
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-audio-buffer-size:
RetroArch [INFO] ::     2048
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-gfxplugin:
RetroArch [INFO] ::     auto
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-gfxplugin:
RetroArch [INFO] ::     auto
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-angrylion-vioverlay:
RetroArch [INFO] ::     disabled
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-filtering:
RetroArch [INFO] ::     automatic
RetroArch [libretro DEBUG] :: set filtering mode...
[... 243 lines cut ...]
RetroArch [libretro INFO] :: GL_RENDERER: Mesa DRI Intel(R) Haswell Mobile
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-vcache-vbo:
RetroArch [INFO] ::     off
Input RomOpen.
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-screensize:
RetroArch [INFO] ::     640x480
GL_ARB_texture_non_power_of_two supported.
RetroArch [INFO] :: Environ GET_VARIABLE mupen64-vcache-vbo:
RetroArch [INFO] ::     off
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 384 frames
RetroArch [INFO] :: ALSA: Buffer size: 1536 frames
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch                                                            /content_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch                                                            /content_video_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch                                                            /content_music_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch                                                            /content_image_history.lpl
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_hi                                                            story.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_mu                                                            sic_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_vi                                                            deo_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_im                                                            age_history.lpl].
RetroArch [INFO] :: [GL]: VSync => on
RetroArch [libretro INFO] :: mupen64plus: Starting R4300 emulator: Dynamic Recom                                                            piler
Segmentation fault
```

gouchi, 2016-09-22:
Thank you for the log.

Try to change [video_monitor_index](https://github.com/libretro/RetroArch/blob/master/retroarch.cfg#L157) in [the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/#using-the-command-line-interface) with
`video_monitor_index = 1`

Don't forget to stop retroarch before making any edits.

Nano shortcuts are CTRL+O (Save), CTRL+W (Search), CTRL+X (Quit)

Mwea, 2016-09-22:
Hi, as I said on my first message, I already tried this, if I try to relaunch retroarch with this command line 
`systemctl start retroarch`
nothing happens

Mwea, 2016-09-22:
After some investigation, I discovered that my computer was not crashing but it was just taking almost 10 minutes to launch a game.. Any idea ?

gouchi, 2016-09-22:
You already tested after changing video monitor index ?

Also you can try to check [which available video outputs ](http://www.lakka.tv/doc/PC/#foxconn-intel-nettop) you have and enable/disable the correct one if necessary.

For the audio, you already make tests with [audio device ](http://www.lakka.tv/doc/Audio-settings/#switching-audio-device-from-the-gui)?

Mwea, 2016-09-26:
Yup as I said, actually my monitor works, audio works too, but launching a game is extremely long on my laptop.
Once the game is launched ( after~10 min) it works well but Launching and Leaving a game take something like 10 minutes

gouchi, 2016-09-26:
Try to change [ribbon options](http://www.lakka.tv/doc/FAQ/#menu-xmb-is-too-slow).

Mwea, 2016-09-30:
Didn't change anything. The laptop used for lakka has 4go of ram and an intel i5 doesn't looks like a lack of performance, it's not lagging, it's just extremely long to load

gouchi, 2016-10-02:
Please try with  `video_monitor_index = 2` in the configuration file. I updated [my message](https://github.com/libretro/Lakka/issues/487#issuecomment-248860706).

Mwea, 2016-10-02:
Same problem.. no changes

Kivutar, 2016-10-04:
OK what about non n64 games?

Mwea, 2016-10-06:
Same for 2048..

hargabyte, 2016-11-13:
Did you ever find a solution for this issue? I am having the same problem. Im using an Intel Nuc with an i5, 8gigs of ram, and SSD but NES and SNES games take 10 minutes to load.

Kivutar, 2016-11-13:
It can be a problem with the texture format used by RetroArch and the graphic card.

What graphic cards are you using?

ClaytonTeve, 2016-11-18:
I actually have this same problem. Games take around 10 minutes to open when I hit run. It just sits on the flower screen until the game eventually starts working. Once the game is open it runs very well. It does the same flower screen for ten minutes when I close a game too. I am reading the documentation currently to make a better issue report based off what my system is doing. I wish I had a spare graphics card to see if its just because of an incompatibility with the integrated graphics on my machine.

hargabyte, 2016-11-18:
Im using an Intel NUC. It has Intel graphics 5500. I was never able to
figure out this issue. I ended up having to install windows and running a
frontend from there.

Kivutar, 2016-12-07:
Duplicate of https://github.com/libretro/Lakka/issues/552

ClaytonTeve, 2016-12-07:
So I actually got mine to work finally, and it has something to do with the settings I changed but because I changed so many I'm not sure exactly what made the difference. To confirm the theory I loaded an default version and got the same long load times issue. Is there any way to post my settings configuration so others having the issue can see the options I used?

levi-lewis, 2016-12-07:
A copy of the settings you changed would be a lifesaver thanks

ClaytonTeve, 2016-12-07:
I changed a lot of settings, and I have no idea what ones did the deed. I was surprised when I turned it on the next day and it loaded up right away. Do you think it would work if I just loaded my entire config folder and you swapped yours out for it?

levi-lewis, 2016-12-07:
We could try. I wont be able to test until i'm on lunch in a couple hours.  By chance have you tried booting more than once since it booted regularly? Mine actually booted regularly once yesterday but then went back to taking 10+ minutes again.

ClaytonTeve, 2016-12-07:
I tested the idea myself by copy and pasting the contents of the configfiles folder. Unfortunately no dice. I wish there was a master list of settings that I could send to you so you could scroll through your settings and change them to the ones I am using. This would also help us test out what setting needs to be changed.

levi-lewis, 2016-12-07:
So you have repeatedly booted with no issues using the new settings? Or was that a one time thing?

ClaytonTeve, 2016-12-07:
On my original one where I changed the settings I have used it several times and it works right away every now time. On my fresh version I run live off of a USB Flash drive I have the same 10 minute flower screen until a successful load problem. Something in the settings I changed fixed it but figuring out what setting did the trick is now what I am working on.

levi-lewis, 2016-12-08:
Ok something very weird is going on. If I get it to the flower splash screen where it freezes and then reboot through SSH and try to immediately run the ROM again it launches without issue.  This only works every other time though so if i try to reboot again it will freeze on the splash screen and then i reboot again and it launches normally.

ClaytonTeve, 2016-12-08:
[RetroArch-1208-055337..cfg.zip](https://github.com/libretro/Lakka/files/638528/RetroArch-1208-055337.cfg.zip)

I found out how to save config files. Hope this helps you.

ClaytonTeve, 2016-12-08:
The file location is Lakka/configfiles/retroarch/config

levi-lewis, 2016-12-08:
Ok thank you i'll take a look later today.

ClaytonTeve, 2016-12-08:
I can confirm this solved it for me. I will take a look and see the differences in the default settings and the ones changed in this config file and maybe we can get to the bottom of whats happening with these intel based systems.

levi-lewis, 2016-12-08:
Just throw the contents of the two files into a thing like diff checker that contrasts two files to check which lines you changed.  That or I will do it once i get home from work.

ClaytonTeve, 2016-12-08:
I made a repository that shows the difference between the starting commit of the default settings and the settings of the fixed version. I just copy and pasted directly into the retroarch.cfg file and it changed everything and fixed the problem automatically. I am not sure what settings made the difference but I hope this helps anyone else with the problem. https://github.com/ClaytonTeve/Lakka-configuration-test

levi-lewis, 2016-12-08:
Thank you so much man.  I just used your config file and it works fine now.  I appreciate it so much.  Not sure what the exact setting was but it works.

Kivutar, 2016-12-08:
Make sure that rewinding is disabled.

Kivutar, 2016-12-09:
So what was it? in the config file?

levi-lewis, 2016-12-09:
I havent had a chance yet to contrast the two files, I will do that this afternoon

Kivutar, 2016-12-09:
You can send both files attached to my email if you have no time to do it

levi-lewis, 2016-12-09:
The updated config is linked up above and I would just be contrasting it to the original config file.

Kivutar, 2016-12-09:
I see a difference in video_threaded

mdelally, 2017-02-14:
I want to confirm that turning 'Threaded Video' to **Off** solved this issue for me. I'm on an old Dell Optiflex 745 with integrated Intel graphics and this fixed it!

closed 2017-02-17 by Mwea

LightShaft, 2017-04-10:
Hello, I have the same problem, but I can't turn 'Threaded Video' to Off, when I unselect the option on the Lakka menu I got stuck on a flower screen, when I switch it to "False" in the cfg file it still "True" when I reboot, I don't understand what I should do (fyi I'm using the latest version on a intel NUC)

Kivutar, 2017-04-10:
Before editing the config file, do a 

    systemctl stop retroarch

After editing, do a

    systemctl start retroarch

LightShaft, 2017-04-10:
Hello, as I said, nothing happened, so I booted on the USB and tried the repair option, nothing happened, I tried a new fresh installation and I've selected the option SSH server and now it works perfectly, I can switch the options I want and the games start (even the arcade games), that's perfect. I don't know if the SSH server was the solution (maybe yes, so I'll say that even if the installer say to install it only if you know what you are doing, install it anyway), but at least it work now and Lakka work. I just don't understand why I can't have a library of arcade games but the rest work pretty well, thank you.

Kivutar, 2017-04-10:
For the Arcade, you need a FB Alpha 0.2.97.39 Split ROMSet

LightShaft, 2017-04-10:
The arcade games run well, I'm just wondering how I can scan the directory to have a list like the lists for the other consoles.

LightShaft, 2017-04-10:
Hello, does this allow me to have a playlist for my arcade games ? 
I found that (https://archive.org/details/FB_Alpha_2016_Reference_Set_-_FBA_v0.2.97.39_ROMs) but it seems to be a big compressed file with "every" game running with FBA, but I want a playlist "Arcade" on Lakka's menu, is this possible ?

Thank you

Sobient, 2017-04-10:
Remember you can get help on offshoot questions or smaller issues on the #lakkatv @ irc.freenode.org so you don't have to go off-topic.

LightShaft, 2017-04-11:
Oh yes you are right please forgive me, I'll check that, thank you.