libretro/Lakka #488: Any means for disabling ssh password authenticantion?

oblitum, 2016-09-21:
OpenELEC [supports this officially](http://wiki.openelec.tv/index.php?title=OpenELEC_FAQ#SSH_Password_change):

> [...] Also you can consider logging in with ssh keys and disabling password logins.

Which I follow on my OpenELEC setups since I can't change user and password. I've tried to figure out how to do it on Lakka but didn't find a way. This is generally done on `/etc/ssh/sshd_config` but read only systems must provide other options, like OpenELEC does through menu configuration (or http://openelec.tv/forum/69-network/39060-prevent-ssh-with-password#39180).

If this is not possible I'd like to make a feature request and try to follow upstream on this.

Kivutar, 2016-09-21:
Ok I don't have a machine to test this now, but you can already try this:

Create a .ssh/authorized_keys in /storage/ and paste your public key in it. Then try to login with your key pair.

I don't remember how to disable password authentication yet.

oblitum, 2016-09-21:
- Copied `/etc/ssh/sshd_config` to `/storage/.config/sshd_config`.
- Edited `/storage/.config/sshd_config` setting `PasswordAuthentication` to `no`.
- Edited (created if non-existent) `/storage/.config/autostart.sh` with:
   ```bash
   #!/bin/sh
   
   /usr/sbin/sshd -f /storage/.ssh/sshd_config
   ``` 
- Disabled SSH in the "Lakka Services" menu entry in the "Settings" menu.
- Reboot.

I'm currently able to login but will have to verify it tomorrow from another machine and close the issue.

Kivutar, 2016-09-21:
Ok good to know. Thanks.

oblitum, 2016-09-21:
Just tried to login from a VM and I was able to login with the root password... I have a suspicion my `autostart.sh` trick is not executing/working. I say this because I was trying to set `TZ` in it too and at first `TZ` just didn't change at all because of it, it only changed when I did `echo "TIMEZONE=America/Sao_Paulo" > /storage/.cache/timezone`.

Any mistake I've done, suggestion, I'd appreciate.

oblitum, 2016-09-21:
Ops, `sshd_config` is in the wrong place. Will check that.

oblitum, 2016-09-21:
Still same problem.

oblitum, 2016-09-21:
I have noticed that it doesn't seem an issue with the `autostart.sh` solely. I have removed the `sshd` call from the script. Then I've reboot and through the Lakka Services menu I've disabled both SSH and SAMBA, I've then saved the current configuration and reboot.

After reboot SAMBA was back as ON in the Lakka Services menu and SSH was OFF, but I was able to login through SSH. I'm running the last cubox-i build. The Lakka Services menu behavior seems inconsistent. It does enable/disable on a given session, but it's not preserved through reboots and displays wrong information.

phatpenguintn, 2017-05-04:
This process did work for me on and Orange Pi PC using Lakka-H3.opipc.arm-devel-20161221192928
I did also make the autostart.sh executable by doing chmod -x /storage/.config/autostart.sh

the TIMEZONE command needs to be:
echo "TIMEZONE=America/Pacific" > /storage/.cache/timezone

lethesdomoi, 2017-05-30:
I also wanted to disable password authentication in Lakka and found a solution which works on the most recent Cubox-i builds, at least.

    echo 'SSH_ARGS="-o PasswordAuthentication=no -o ChallengeResponseAuthentication=no -o KbdInteractiveAuthentication=no"' >> /storage/.cache/services/sshd.conf

Reboot or restart ssh. I probably don't need to say that before running that command you should be absolutely sure that you have copied the public ssh key for the computer(s) you want to have access to the Lakka box in /storage/.ssh/authorized_keys.

oblitum, 2017-05-30:
@lethesdomoi many thanks, I'll try it next time I get to my Lakka Cubox-I.