libretro/Lakka #490: Odroid XU4 - Bluetooth inactive after reboot

epsilon3379, 2016-09-24:
Hi :)

I cannot succeed to keep the bluetooth active after a reboot on the Odroid XU4.... even with the following command :

nano .config/udev.rules.d/bluetooth.rules

with this content :

# Set bluetooth power up
ACTION=="add", KERNEL=="hci0", RUN+="/usr/bin/hciconfig hci0 up"

After a reboot, the "systemctl status bluetooth" return "Active: inactive (dead)"

I always have to type "systemctl start bluetooth" and then,  my gamepad is connected automatically.

Any ideas ?

Thanks

epsilon3379, 2016-09-28:
Finally found a solution/workaround for the "bluetooth inactive after reboot" problem !

Here's the solution in 4 steps :

1. Enable the autostart.sh file
`systemctl enable retroarch-autostart.service`

2. Create & open the autostart.sh file
`nano /storage/.config/autostart.sh`

3. Fill the file with this content
```
(
 systemctl start bluetooth
) &
```

4. Restart the Odroid
`shutdown -r now`

And voilà

The bluetooth service should now start automatically after reboots

Kivutar, 2016-10-04:
Ok thanks for this information. I need to check why the service doesn't automatically start when bluez.conf is found.

neher, 2016-10-15:
On the Pi3, I have a similar problem (clean install version 5 oct. 2016).

When enabling bluetooth using systemctl, it refuses to start because of the missing bleuz.conf file (which is cleared on every reboot :( )
When I just enable the bluetooth services trough the lakka GUI, everything just works. The bleuz.conf file is being created at startup.

epicbrew, 2017-06-23:
I've dug into this a bit and it seems the only problem is that systemd is creating the wrong link when `systemctl enable bluetooth` is issued. It's creating the bluetooth.service link in `.config/system.d/bluetooth.target.wants` instead of in `.config/system.d/multi-user.target.wants`. Manually creating the correct link enables the service properly throughout reboots:

    cd .config/system.d/multi-user.target.wants
    ln -s /usr/lib/systemd/system/bluetooth.service ./bluetooth.service

Interestingly, if you disable the bluetooth service, systemd notices the manually created link and deletes it in addition to the link it created in .config/system.d/bluetooth.target.wants.
I'm running Lakka 2.0 on the XU4.

Kivutar, 2017-06-23:
OK, thank you for this helpful information. I'm going to check what's different in the XU3 project that could cause this.