libretro/Lakka #491: Correction to Logitech F710 udev layout correction

wboyo, 2016-09-28:
The current controller for mappings for the Logitech Gamepad F710 that can also operate as a Logitech Cordless RumblePad 2 have the default RetroPad YXBA maps defined wrong as the actual physical button order is arranged in a XYAB format.  That attached to cfg files correct this issue.  Also attached is an new udev cfg for an QuanBa 2in1 Arcade Stick.

[Logitech Gamepad F710.cfg.txt](https://github.com/libretro/Lakka/files/498016/Logitech.Gamepad.F710.cfg.txt)

[Logitech Logitech Cordless RumblePad 2.cfg.txt](https://github.com/libretro/Lakka/files/498018/Logitech.Logitech.Cordless.RumblePad.2.cfg.txt)

[ACRUX QuanBa Arcade JoyStick 1008.cfg.txt](https://github.com/libretro/Lakka/files/498023/ACRUX.QuanBa.Arcade.JoyStick.1008.cfg.txt)

gouchi, 2016-09-28:
Thank you for your contribution !

Is it possible you add [label](https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/OUYA%20Game%20Controller.cfg#L28) ? And if possible [vendor/product id ](https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/8Bitdo_Classic_FC30_USB.cfg#L6)?

To get vendor/product id, you need to type `lsusb`  (via [Lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/)) to list all devices attached via USB then get the ID XXXX:XXXX you need to convert it from hexadecimal to decimal.

wboyo, 2016-09-30:
All udev/.cfg's updated and tested.  Should have everything you asked for now!

L3 & R3 both mapped to the extra "Home" button found on the QuanBa stick to allow exiting cores in game.

[ACRUX QuanBa Arcade JoyStick 1008.cfg.txt](https://github.com/libretro/Lakka/files/502377/ACRUX.QuanBa.Arcade.JoyStick.1008.cfg.txt)

[Logitech Gamepad F710.cfg.txt](https://github.com/libretro/Lakka/files/502378/Logitech.Gamepad.F710.cfg.txt)

[Logitech Logitech Cordless RumblePad 2.cfg.txt](https://github.com/libretro/Lakka/files/502380/Logitech.Logitech.Cordless.RumblePad.2.cfg.txt)

Many, many thanks to the Lakka & RetroArch team (not forgetting all the emu devs, rom dumpers and even shader scripters).

gouchi, 2016-09-30:
Thank you [your contribution](https://github.com/libretro/retroarch-joypad-autoconfig/pull/235) has been merged ! It will be available in the next release of Lakka !

closed 2016-09-30 by gouchi