libretro/Lakka #492: Stuck in Flower Screen

suteriosu, 2016-09-29:
As mentioned by other people, I can't seem to get Lakka to work on my personal computer.
Downloaded the latest nightly and made the bootable usb. Works fine on my laptop and older PC, but refuses to work on my main PC.
I want to have the usb as a live Lakka device, to work on all my devices. So I am not talking about clean install.

gouchi, 2016-09-30:
Is it possible to provide some [logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/) (Lakka sytem and graphic card logs) ?

Thank you.

suteriosu, 2016-09-30:
I can do that. But how do I access the terminal from the flower screen? (noob here)

gouchi, 2016-09-30:
Please check howto to [access Lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/).